prusa3d/PrusaSlicer #7503: Wipe tower GCode has double spaces

fpikus, 2021-12-16:
### Version
_Version of PrusaSlicer used goes here_
2.3.3
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Lotmaxx Shark V1
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Generated GCode has double spaces in wipe tower sections. 
Shark ignores lines with double spaces (yes it shouldn't, but ... Chinese firmware ... )
Example:
; CP EMPTY GRID START
; layer #2
G1  X88.000 Y90.875 
G1  Y91.375 
G1  Y78.875  E0.4751 F2900
G1  X148.000  E2.2804
G1  Y91.375  E0.4751
G1  X88.000  E2.2804
G1  X88.500 
G1  Y90.875 
G1  Y79.375  E0.4371
G1  X147.500  E2.2424
G1  Y90.875  E0.4371
G1  X88.500  E2.2424
_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[Bi-color calibration cube_0.2mm_PLA_SHARK1_0.4.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7724525/Bi-color.calibration.cube_0.2mm_PLA_SHARK1_0.4.3mf.zip)

lukasmatena, 2021-12-16:
You are correct. It is like this for a really, really long time, I see some double spaces in 2.0.0. This should be fixed, but not now before 2.4.0 release. The G-Code should be valid, it is just a bit longer than it needs to be.

fpikus, 2021-12-16:
"Should" is an operative word. Prusa printers accept it just fine. Lotmaxx printers ignore all lines with double spaces so I have to edit GCode in VIM every time.

lukasmatena, 2021-12-16:
@fpikus Well, according to https://tsapps.nist.gov/publication/get_pdf.cfm?pub_id=823374:
```
Spaces and tabs are allowed anywhere on a line of code and do not change the meaning of the
line, except inside comments. This makes some strange-looking input legal. The line “g0x +0. 12
34y 7” is equivalent to “g0 x+0.1234 y7”, for example.
```
Unless i misunderstand something, the G-Code should be valid and if some firmware rejects it, it is its fault and I would report it to the author. We will remove the extra space in PrusaSlicer (it is not needed), but not before 2.4.0.

bubnikv, 2021-12-16:
```
G1  X147.774  E2.2424 F1584
```

is a perfectly valid G-code. It is true that we should not emit double spaces, but as @lukasmatena noted, we need to release PrusaSlicer 2.4.0 thus we will fix it in 2.5.

Please notice your 3D printer vendor to fix their firmware for now.

bubnikv, 2021-12-16:
We are not going to fix it for PrusaSlicer 2.4.0. We will fix it after the release.

You can write yourself a simple post processing script, even a simple

```
sed 's/  / /g'
```

will likely work

fpikus, 2021-12-16:
I know it's valid. It's just you guys have your act together, and if Lotmaxx knew how to write non-crap firmware, this wouldn't even be the top problem on their list. I understand that 2.4.0 has to be released now. I was hoping for 2.4.x.

DeeKay64, 2022-02-07:
Well, I've reported this bug (and many others!) when the printer had basically just been released and I did my first slices in Prusaslicer. I think I was the first one to figure out what was happening. But -as many other bugs- it never got fixed, and it seems we won't be getting more firmware updates for the printer, so it would be great if you guys could fix this on your end, even though the Gcode *should* be valid with double spaces, there's really no reason to use them in the purge block code! ;-)

DeeKay64, 2022-02-07:
It should be pointed out that some arguments even have *triple* spaces infront of the E value!...

lukasmatena, 2022-02-08:
PrusaSlicer 2.4.1 allows to define G-Code substitutions. You can use regular expression `[ ]{2,}` to match any number of spaces higher than one and replace them with a single space. The repeated space in the wipe tower code will not be fixed before 2.5.0. The G-Code is composed in several steps and it is internally post-processed later, there is a risk that a space will become missing in some scenario if the fix is not perfect. It is not an issue for any decent firmware. Also, I suggest to not buy printers you yourself describe as crap. There are other solutions on the market, for example I heard Prusa is not bad.

lukasmatena, 2022-02-09:
Could you please try the following build? It might be fixed, but I'm not sure what else I broke:

[macOS-lm-wipe-tower-base-and-spaces-gaab92df84](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.5.0-alpha0+11-lm-wipe-tower-base-and-spaces-universal-asan-gaab92df84-202202091418.dmg)
[Linux-lm-wipe-tower-base-and-spaces-gaab92df84](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.5.0-alpha0+11-lm-wipe-tower-base-and-spaces-linux-x64-asan-GTK2-gaab92df-202202091410.AppImage)
[Windows-lm-wipe-tower-base-and-spaces-gaab92df84](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.5.0-alpha0+11-lm-wipe-tower-base-and-spaces-win64-gaab92df84-202202091410.zip)

user: `prusaslicer`, password: `slicer`, if you are asked about it. Thanks.

fpikus, 2022-02-10:
Could not get far enough.
1) When started, it shows a dialog box:
[image: image.png]

2) After I click OK, the main platter window is empty, no bed, nothing. I
tried blindly dropping a model there, got this:
[image: image.png]

I can close it but without anything drawn in the window, I have no idea
what is happening.

lukasmatena, 2022-02-10:
@fpikus Thanks. I do not see your images. The version I sent here loads its configuration from `PrusaSlicer-alpha` folder. If you can still run the 2.4.0, there may be something rotten in the configuration. Either that, or some other change caused it. Either way, it's worth investigation. Could you please:
- upload the images that are missing in your previous post
- go to `(home)\AppData\Roaming` and rename the `PrusaSlicer-alpha` folder to something else, such as `PrusaSlicer-alpha-backup`
- check if you now can run the build I sent you (aab92df84)
- if you can run it, it would be great if you could zip the folder you backed up and send it to lukas.matena (a) prusa3d. cz  for inspection

Thanks.

fpikus, 2022-02-11:
[image]
[image]

Dropped the files directly into Github, hopefully, this works better than the email gateway.

I removed the PrusaSlicer-alpha folder but I still cannot run the build, I get exact same errors. I tried to answer Yes or No to importing configuration (removed the alpha folder every time), it makes no difference to the errors. The wizard will run if I don't import, and it has the usual choices of printers, but no images of the printers. Instead, there is a gray rectangle with text NO IMAGE.

lukasmatena, 2022-02-11:
@fpikus Interesting. The error with missing shaders would happen if you used old resources directory and the new binary, that is, if you only unpacked the binaries and placed them alongside the old resources. Anyway, to be save, here is a Windows build that is based on 2.4.1-beta1, with no other development but my wipe tower fix:
[Windows-lm-wipe-tower-base-and-spaces-g618ae6ade](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.4.1-beta1+179-lm-wipe-tower-base-and-spaces-win64-g618ae6ade-202202111024.zip)

fpikus, 2022-02-11:
I unzipped the archive in its own directory and ran from there, so
everything that was in the zip file is unpacked together.
I tried the new build, got "Unhandled unknown exception; terminating the
application." then "Internal error: Could not load bitmap." There were
"details" on the second box:
10:19:10: can't open file
'E:\Downloads\PrusaSlicer-2.4.1-beta1+179-lm-wipe-tower-base-and-spaces-win64-g618ae6ade-202202111024\resources\icons\PrusaSlicer.png'
(error 2: The system cannot find the file specified.)
10:19:10: Failed to load image from file
"E:\Downloads\PrusaSlicer-2.4.1-beta1+179-lm-wipe-tower-base-and-spaces-win64-g618ae6ade-202202111024\resources\icons\PrusaSlicer.png".
10:19:10: Internal error: Could not load bitmap: PrusaSlicer

lukasmatena, 2022-02-14:
@DeeKay64 @f1vefour Were you guys able to run and test that build? Thanks.

@fpikus 
> 10:19:10: can't open file
'E:\Downloads\PrusaSlicer-2.4.1-beta1+179-lm-wipe-tower-base-and-spaces-win64-g618ae6ade-202202111024\resources\icons\PrusaSlicer.png'

Does that file exist in that location or not?

fpikus, 2022-02-15:
All files starting with "p" and later were missing. When I unzipped the
file again, I saw why: Windows Defender demanded a cloud scan of some file,
I must have refused the first time so it aborted extraction.

Now it runs and the tower gcode is fine, I checked with vim and did a
search, no double/triple spaces anywhere.

It does have the error about demanding G92 E0 in layer change code, but I
assume it's a feature.

Fedor

lukasmatena, 2022-02-15:
@fpikus Thanks. Yes, that is a feature to prevent loss of precision in printer fw with long prints.

f1vefour, 2022-03-26:
Thank you @lukasmatena for the build, I apologize for not responding sooner. I concur with fpikus that it fixes the issue.

lukasmatena, 2022-03-30:
Thank you both for the testing. The fix is now in master for the 2.5 release. Closing.

closed 2022-03-30 by lukasmatena

DeeKay64, 2025-01-17:
I just checked in PS 2.9 - and the double/triple spaces are gone, not a single one is found, great! ;-) Thanks, guys!