prusa3d/PrusaSlicer #7504: Missing Bottoms of objects, Bridging printed on build surface after Re-slice

Snuff1eupagus, 2021-12-16:
### Version
PrusaSlicer-2.4.0-rc+win64-202112141553 

### Operating system type + version
Win 7 Pro 64 SP1 & Win 8.1

### 3D printer brand / version + firmware version (if known)
MK3 Series

### Behavior
* _Describe the problem_
Please see images in "Steps needed to reproduce the problem"

* _Steps needed to reproduce the problem_

Right click [Add shape] / Box. Right click on Box [Add settings] / Skirt and brim :Brim width, set to 5mm (or what ever you like).

Right click on Object [Add Part] / Cylinder, in the [Part manipulation] pane Scale Cylinder to 10%. Drop to bed (using Icon).

[image]

Select Shape-Box From the Subobjects in the sidebar,

[image]

Right click select "Infill" do the same for "Layers and Perimeters" select the resulting properties and set all to Zero, 

[image]
Hit the "Slice now" button. 

Initial result. I might add I was hoping for there to be a Brim printed around an invisible box as before.
[image]

With the properties for the Shape-Box subobject still selected add 1 or more Bottom: layers, and Re-slice.

Final Result.
[image]

You should see something like the image above, only bridging for bottom layer on the build surface & no brim.
The Objects don't appear to be elevated, and the using the "Drop to bed" Icon has no effect on the Object or it's 
Subobject.
Restoring the settings to default did not restore the bottom layer or brim, I tested this on the Box, Cylinder, and the Bunny. 

I ran this process 20 times +, with the same results before Submitting. 

Notes: Models such as the Cylinder left artifacts when adding the bottom layers back in.
These are not just visual artifacts, they appear in the G-code.
I went back to 2.4-Beta 4 and got same results, I did no farther testing.

artifacts 2
[image]

artifacts no bottoms, no brims.
[image]

* _Expected Results_ 
* 
Result I should have gotten. 
[image]

Expected Results when adding the defaults back to the properties.
[image]

* _Actual Results_

Please see images in "Steps needed to reproduce the problem"

#### Project File (.3MF) where problem occurs

More Notes: This file is ready to go, I setup the "Steps needed to reproduce the problem" Just Slice and then add 1 or more Bottom: layers, in the properties mentioned and Re-slice.
I experimented with this file as well, If you change the properties of one of the Objects before the first time slicing, That object will print correctly the remainder of the time.

[Missing Bottoms.zip](https://github.com/prusa3d/PrusaSlicer/files/7724493/Missing.Bottoms.zip)

Sorry about the long post, I was trying to be through.

lukasmatena, 2021-12-16:
Thanks. I can see there is something wrong. The outcome seems to really depend on whether it was sliced before or not.
I tried in 2.3.0 but could not reproduce it. I tried in 2.4.0-beta1 and it crashed. We'll look into it.

bubnikv, 2021-12-16:
There seems to be two diffferent issues:

1) Brim is sometimes calculated around the "invisible" object, sometimes not, depending on whether one sliced after each operation or not.

2) Spurious floating extrusions, which seems to be caused by the "ensure vertical wall thickness" feature. If you switch the feature off, the spurious extrusions disappear.

Both are edge cases, I don't think it is worth fixing for 2.4.0 release.

bubnikv, 2021-12-16:
I think the issue with the brim not being calculated for the invisible object is caused by the invalidation logic, which does not expect the object to be not printed at all.

Frankly I tend to either stop slicing with an exception explaining that such a setting does not make sense, or to consider such an object to be hidden.

Snuff1eupagus, 2021-12-16:
I use this, its a powerful tool, a method to achieve results in slicer that are not avalible otherwise, just one example of this was posted in Prusa Forums in the alpha_3 release announcement. 
 
https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-2-4-0-alpha3-is-out/

Scroll down a short way and you will see a post by @swiss_cheese that explains (briefly) one way that this is needed, I believe it's worth fixing, I use this process often not only for the purpose described in that post, which is only newly available since the addition of Skirt & Brim to modifiers, but for several other things as well prior to that addition that I'm not prepared to mention here, that are impotent to work flow and the construction of meaningful project files.

I'm a bit, concerned to hear that the things that have been making slicer work for me over the years appear to be viewed not worth fixing and that the usage doesn't seem to be realized. The reason I even tested this scenario is that it's one of my main use cases, It offers so much functionality well beyond the example provided. 

Regards

bubnikv, 2021-12-17:
I don't see any comments by Swiss_Cheese on empty objects. The use of empty objects is out of specification.

Snuff1eupagus, 2021-12-17:
The post was about using multiple Brims with different specifications on a single model, as well as only applying a brim to a selective area of a model (not the entire model).

It was done by duplicating the "parts" of the mesh that needed the brim, then loading them together with the whole model "A Dragon model was used to demonstrate", after splitting to objects (Dragon feet in this case), a user would apply modifiers, with "skirt and brim", "infill", and "layers and perimeters" to each of the (4 foot objects). The "infill" and "layers and perimeters" would be set to "Zero" so no double printing will occur in those areas, and the brim settings for each of the foot objects could be set independently of one another while allowing for a proper hold the brims printed perfectly, the empty objects only acted as a place holder for the brims on each foot.  In the example used one foot is left un-brimed. This application/method/work around, whatever you want to call it, is far reaching, I use it in several different ways in slicer, and it's worked great till now. There is no other reasonable way available to do this currently.

closed 2021-12-18 by Snuff1eupagus

bubnikv, 2021-12-20:
> It was done by duplicating the "parts" of the mesh that needed the brim,
then loading them together with the whole model "A Dragon model was used to
demonstrate", after splitting to objects (Dragon feet in this case), a user
would apply modifiers, with "skirt and brim", "infill", and "layers and
perimeters" to each of the (4 foot objects). The "infill" and "layers and
perimeters" would be set to "Zero" so no double printing will occur in
those areas, and the brim settings for each of the foot objects could be
set independently of one another while allowing for a proper hold the brims
printed perfectly, the empty objects only acted as a place holder for the
brims on each foot. In the example used one foot is left un-brimed. This
application/method/work around, whatever you want to call it, is far
reaching, I use it in several different ways in slicer, and it's worked
great till now. There is no other reasonable way available to do this
currently.

Thanks. We may manage to implement some way to configure brims on object
islands in PrusaSlicer 2.5, but I cannot promise.

Snuff1eupagus, 2021-12-21:
Well, this wasn't a new feature request, frankly I think just fixing the exception would be fine, Ive been doing this for years aside from the brims for many reasons, and never had a problem with the work around till now.
I was simply trying to alert you that something isn't right. 

but if your not going to fix it, your not going to fix it.

I'll find another way.

Thanks