prusa3d/PrusaSlicer #7505: G1 E GCODE is getting generated without a whole number

cjackson234, 2021-12-16:
### Version
2.4 RC

### Operating system type + version
Windows 10 19042.1348

### 3D printer brand / version + firmware version (if known)
Voron Switchwire - Klipper v0.10.0-184-gdd714fc7

### Behavior
G1 E GCODE is getting generated without a whole number
For instance 
G1 E0.8 F2100 outputs as G1 E.8 F2100

* _Steps needed to reproduce the problem_
Slice any file
* _Expected Results_
G1 E0.8
* _Actual Results_
G1 E.8
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_
no

neophyl, 2021-12-16:
https://github.com/prusa3d/PrusaSlicer/issues/6996

lukasmatena, 2021-12-16:
@cjackson234 Does it cause any issues? Does your Klipper+Voron combination not process it correctly?

cjackson234, 2021-12-16:
Klipper immediately throws an error and the print fails

cjackson234, 2021-12-16:
> If that is the case, would you please open a Klipper issue?

Sure, but they will correctly point out that the problem doesn't exist in other slicers.

bubnikv, 2021-12-16:
But they don't conform to G-code standard.

cjackson234, 2021-12-16:
I will reach out to them now.

Is this something that could be patched in the meantime by adding an additional gcode flavor?

bubnikv, 2021-12-16:
We are not going to fix it for PrusaSlicer 2.4.0. We will fix it after the release.

You can write yourself a simple post processing script, even a simple 
```
sed 's/  / /g'
```
will likely work

cjackson234, 2021-12-16:
I will test that while I wait on their response. Thanks

th33xitus, 2021-12-16:
> We are not going to fix it for PrusaSlicer 2.4.0. We will fix it after the release.

@bubnikv 
Why did you change that behavior in the first place? No criticism, just wondering.

bubnikv, 2021-12-16:
> Voron

Sorry guys, this response was meant to  https://github.com/prusa3d/PrusaSlicer/issues/7503

bubnikv, 2021-12-16:
> Klipper immediately throws an error and the print fails

Which klipper.

Frankly I am printing with PrusaSlicer on Prusa Mini with Klipper and I did not experience any issue.

bubnikv, 2021-12-16:
@th33xitus 
> Why did you change that behavior in the first place? No criticism, just wondering.

We optimized G-code export, you will enjoy a significantly faster G-code export now. Along the way we optimized for the file size while conforming to the G-code specification. I am frankly pretty surprised if there was a number parser that cannot parse such a number.

cjackson234, 2021-12-16:
Klipper v0.10.0-184-gdd714fc7

What version are you using?

th33xitus, 2021-12-16:
@bubnikv Sorry, yeah i saw that in the release notes of 2.4.1-alpha2, that's why i deleted my comment. Should have looked first. 

Im just printing g-code generated by the new 2.4.1-rc without any issues. My current Klipper version is `v0.10.0-169-g520273e5`.

cjackson234, 2021-12-16:
> @bubnikv Sorry, yeah i saw that in the release notes of 2.4.1-alpha2, that's why i deleted my comment. Should have looked first.
> 
> Im just printing g-code generated by the new 2.4.1-rc without any issues. My current Klipper version is `v0.10.0-169-g520273e5`.

Very interesting. Would you mind sharing additional details? Are you using octoprint or fluidd/mainsail?

th33xitus, 2021-12-16:
Im using Mainsail.
I will update my Klipper to newest and try again.

bubnikv, 2021-12-16:
I am using
https://github.com/raymondh2/MainsailOS
https://github.com/matthewlloyd/klipper.git
the klipper for Prusa Mini was synced with upstream around 5 months ago.

th33xitus, 2021-12-16:
I just updated to Klipper to `v0.10.0-184-gdd714fc7`.
Working fine... no issues on my side. Same g-code as before.

cjackson234, 2021-12-16:
Closing the issue for now as other users were not able to reproduce. I'll post back here when I find a solution.

closed 2021-12-16 by cjackson234

cjackson234, 2021-12-16:
This is a configuration issue. How variables set has changed. In my print start gcode was parsing a value incorrectly. That caused all subsequent G1 E commands to fail. Oddly, manually changing the compiled gcode to start with a whole number fixed the issue for me. 

That being said, thank you for looking into this.

ledvinap, 2021-12-17:
You can add zeroes with
```
sed -e 's/\([eE]\)\.\([0-9]\)/\10.\2/g'
```

kevinmichaeldudley, 2021-12-22:
My "day job" is developing and marketing CNC software for 35 years.
I have over 1000 users running "god knows how many different CNC machines".
FYI: The fact is, in the CNC world, not all machines will accept "no leading 0".
Is it "not to standards" possibly, but in the real world we make OUR software to WORK on everything, as I believe the slicer gcode output should also.
JMHO