prusa3d/PrusaSlicer #7506: Cannot manually add supports

jess1key, 2021-12-16:
Prusa 2.4.0

Windows 10 64

Hi im very inexperienced with printer knowledge - but I have run into an issue that i cannot understand. Im using the newest version of prusa slicer 2.4 and now since its been installed I can no longer add manual supports. I do not know why this new version isnt allowing me to manually add supports, I have tried everything I can think of. Please be aware im not very technically savvy, and this has been most frustrating. 

This happened once before when I tried to update to the newer version of prusa slicer and at that time i was able to uninstall the new version, and go back to the older version. Any help would be appreciated. I barely know any of the lingo here so please keep that in mind when instructing how to fix this problem. 

I would like to just re-install the old version I was using, and now it does not appear i am able to do so.

lukasmatena, 2021-12-16:
Could you please described the issue in more depth? What version are you using exactly? 2.4.0-rc? 2.4.0-beta4? Something else? Could you post a 3MF so we can try to reproduce it with your config? Are you talking about SLA supports or FDM supports? What does it mean that you "can no longer add manual supports"? What happens if you try? Does it not do anything? Does it crash? Does it report something? Are they added but do not appear in the output? Etc.

I just quickly checked SLA and FDM supports in 2.4.0-rc and see no issue.

bubnikv, 2021-12-16:
Maybe you are in a simple mode thus FDM paint on supports are suppressed?

jess1key, 2021-12-27:
I'm using 2.4 and the only thing I can do is allow the slicer program to auto generate supports. If I click on manual editing it only allows me to add supports in specific spots, if at all. Sometimes when I click to add the supports it places the supports floating in air, attached to nothing. There are spots that I need to manually add supports or else my print will fail. Now I cannot add them at all. Someone else posted about this and the case was closed. This seems to happen with some files and not others.

bubnikv, 2022-01-03:
Are you using PrusaSlicer for SLA or FDM prints?

closed 2026-09-29 by prusaslicer-bot[bot]