prusa3d/PrusaSlicer #7509: Feature request: Change order of Presets categories in Settings dropdowns

mix579, 2021-12-16:
### Version
Any

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
New feature request:

Currently, the order of entries in the Print settings, Filament settings, and Printer settings dropdowns is System Presets, User Presets, and—in the case of Printer Settings—Physical Printers.

Given that users create user presets because presumably they want to use those, not system presets, it usually involves a lot of unnecessary scrolling to go past the system presets to the presets you actually want to use.

A simple way to make life easier is to reverse the order to Physical Printers, User Presets, and finally System Presets.

mix579, 2022-02-02:
I realized I submitted it before: https://github.com/prusa3d/PrusaSlicer/issues/6820

rtyr, 2022-05-16:
Closing as a duplicate of #6820.

closed 2022-05-16 by rtyr