prusa3d/PrusaSlicer #7510: Error message when generating the g-code

mihai-stanimir, 2021-12-16:
### Version
_PrusaSlicer 2.4.0-rc_

### Operating system type + version
_Windows 10_

### Behavior
I get this error message:

[image]

Steps:
* Download, unzip, and run PrusaSlicer-2.4.0-rc+win64-202112141553
* Load .3mf file generated from Fusion 360 as independent bodies (same file is open in PrusaSlicer 2.3.1 and works fine)
* Click arrange
* Switch to preview
* _Actual Results_ I get an error message. See screenshot above.

_Is this a new feature request?_
No

bubnikv, 2021-12-16:
Are you using any post processing script? Do you have windows explorer
opened at that temp directory?
Something somewhere may be locking the file that cannot be renamed.

bubnikv, 2021-12-16:
windows explorer has a weird ways of opening files for preview.
Antiviruses may do the same. They watch folders for changes and they may open a newly created file for inspection.

n8bot, 2021-12-16:
Please don't get mad at me for making this suggestion, it's part of me learning about all of this:

Could PS quietly retry several times before alerting the user, in the case of encountering that error (which obviously PS already handles)?

n8bot, 2021-12-16:
Oops. Nevermind. Looking at the code, the comment indicates that it should retry. Sorry.

bubnikv, 2021-12-17:
There is a difference between Windows and Unix based / Unix like systems (read OSX and Linux): While Unix like systems let one delete or move a file while being opened, Windows does not. Unix handles that situation by reference counting the files referenced from the file system AND from all the open file handles. The file is being deleted first after the reference counter drops down to zero. Windows does not do that, which makes things complicated.

We have copied a large chunk of code from clang compiler suite for safe moving / renaming files in a cycle. We are not the only one affected by this Windows design quirk.

n8bot, 2021-12-17:
Yeah, I encounter it a lot with git for windows when trying to revert or rebase or undo changes when the file is being fiddled with by intellisense or something. Thank you.

bubnikv, 2021-12-20:
> It isn't happening anymore, so I can't get any more info.

Let's close the issue then. Please reopen if it happens again.

closed 2021-12-20 by bubnikv

dbgitter, 2024-01-04:
I am getting a very similar error message:
"PrusaSlicer has encountered an error Renaming of the G-code after copying to the 
selected destination folder has failed. Current path is /Users/db/Desktop/PrusaLackTube06_0.05mm_PLA_MK2S_2h55m.gcode.tmp. Please try exporting again."  
 
Using iMac with macOS Mojave v 10.14.6 and PrusaSlicer 2.7.1+MacOS-x64    
No post processing   
   
The error is intermittent. 
Cura has no problem working with this file/folder.

Question: 
   It does generate a .tmp file which appears to have valid content.  Should I just be able to rename it to .gcode and go ahead and use it?