prusa3d/PrusaSlicer #7512: My prints are too short

MikeSavad, 2021-12-17:
I an brand new to printing and this program. I used some good settings from cura to put in here since there is no direct translation to import them.

I'm using a twotrees blu-5 (bluer plus), using my settings.

When I print, the height is wrong. On average its 3/4's high as it should be. So like I printed a tiny cube:

https://www.thingiverse.com/thing:170922

I like this because its pretty fast.  It is supposed to measure 15mm high. But I got only 12 or so. But the weird thing is, its proportional. If I didn't know it had measurements I would say its printing fine.  So like the edges are 5mm, mine is 3. The base is 2mm, mine is 1.5mm.

This makes making things hard.

I thought it was the printer, adjusted everything I could think of, then tried it in cura. And that worked perfectly. 

I don't know how to use this program well enough to know how to fix it. Its probably a simple fix, maybe something I bumped into, though I can't imagine there being a scaling function. 

Any ideas? Is it a bug in the software?

The bios in the printer is a robin nano 3 I think with a makerbase v1.03 (i'm too afraid to brick it to upgrade that). I have it set to marlin flavor, tried it in repetier, same issue. Tried it in maker base and the printer went crazy. Terrible first layer and the extruder didn't know which way to turn.

Thanks

----Mike Savad
 

### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Alpacad, 2021-12-17:
@MikeSavad does the preview show the correct height in the slicer?
If it is, maybe that's not a software isssue, but rather your z motors step/mm need to be calibrated.
I don't know if Cura overrides your axis step/mm to make it print with dimensional accuracy, so it might me an idea to check your step per mm values.

### **The correct way:** 
[use prusa step per mm calculator](https://blog.prusaprinters.org/calculator_3416/) and check if the value is correct by sending a M503 command to the printer, look the Z value under M92

**If it's not the same value**, you can add M92 Z[value] to your start G-code, or store it directly in your firmware by sending M92 Z[value] to your printer and then M500 to save it.
Do the same for the other axes if needed and see if it prints with the correct dimensions.
**If it's the same value**, then, might be a software issue.

MikeSavad, 2021-12-17:
I can't calibrate the motors because they gave me firmware that doesn't have that ability. But in cura it printed fine. I've been blaming it on the machine all this time, but the software seems to be at fault.

But I don't know how to use either program that well. 

That calculator looks neat, but i'm not sure about what to fill in there. The printer came pre built, so I don't know the specifics.  I'm also not sure how to send codes to the printer (I got the printer like a week ago and i'm taking  a crash course now). 

There wouldn't happen to be a video how to do this? I'm more visual. Thanks

----Mike Savad

MikeSavad, 2021-12-17:
One thing i'm thinking is, that I couldn't find a printer profile for prusaslicer. But did find one for cura.  (Or at the very least I don't know hot to install the settings into prusa. 

In cura there is something called print head settings. I don't know if this is a thing I can enter into prusa.

My first print was an XYZ cube that came out so bad, that it looked like I made it out of fudge and left it in a window to melt... But the Z was OK, better than now anyway. It was a cube. Not a rectangle. So i'm thinking that I must have messed up something by putting in a wrong number somewhere.  I'm just glad its not the machine, but it opens a new mystery up. 

I was hoping to find that info you asked about in cura, but so far I only found that print head settings.

----Mike Savad

lukasmatena, 2021-12-17:
@MikeSavad
> the software seems to be at fault

It is IMHO quite unlikely that PrusaSlicer would simply scale the model without provocation, not to mention that the only person reporting this claims that he does not know the software well, that he has no profile made for the printer and that other prints had issues too. It is much more likely that there is something in your Printer Settings (Custom Start G-Code maybe) that causes the issue. I would consider it certain in fact. I suggest to get a PrusaSlicer profile for your printer first (Cura profiles are not compatible).

Either way, there is by far not enough info here. We would need your PrusaSlicer and OS version and a 3MF to reproduce the behavior. But I'd like to say in advance, that this is PrusaSlicer issue tracker, meaning that after a bug in PrusaSlicer is ruled out (as I assume), nobody can troubleshoot the problem for you.

MikeSavad, 2021-12-17:
I'm not blaming the software per se, more like my settings. That maybe I entered something wrong. Because the hardware works I have to go to the next thing that may be the problem. 

It would be great if there was a profile for my printer, I did look, there is one for another model but not this exact model.
[CubeCircle.zip](https://github.com/prusa3d/PrusaSlicer/files/7735731/CubeCircle.zip)

I included the file i'm working with right now. 

My guess is, its a simple fix that is beyond me. 

I don't think its a bug but maybe?  Someone said in the past (i'm still trying to find where I saw it), that he had issues with this slicer over other ones and it came out short too.  Which is why I had the idea to try it in cura as a check.

----Mike Savad

rtyr, 2021-12-17:
The output from PrusaSlicer is OK, last layer is printed at Z14.95 as expected.

I guess your machine's limits are set incorrectly, which results in skipping steps during Z travels.

Set this settting to "Ignore" or "Use for time estimate", reslice the g-code, restart the printer and try again.

[image]

MikeSavad, 2021-12-17:
Its worth a shot. I'll make one up and see how that goes. Thanks.

Is there a way to find out what the machine limits are?

----Mike Savad

MikeSavad, 2021-12-17:
Sadly no. I didn't do the ignore, not sure If that would make a difference or not. This was short, but weirdly, the center was all melted. Where as the others still had their original form, this one, the top and middle were flattened. Like how a building pancakes on the way down.  On this one, it was not only short through the first step, but the long post, its like the Z never moved up there. 

----Mike Savad

rtyr, 2021-12-17:
> Is there a way to find out what the machine limits are?

You should use the same values as set in your firmware (those values are usualy visible after sending M503 command via console). It is better to set it to "ignore" if you are not sure what to set. You can also try to set slower Z travel speed in print settings (for example 5mm/s). 

Anyway, it is not PrusaSlicer related problem, so I am afraid I have to close this issue.

Your g-code printed on different printer:
[image]

closed 2021-12-17 by rtyr