prusa3d/PrusaSlicer #7514: Feature Request: IDEX unused extruder cooldown

schuessm, 2021-12-17:
Hello,
I have a feature request for my IDEX printer:

When the second/first extruder is not in use anymore, or generally, at a print it must cool down or should not head up. The problem can be that this otherwise can block the nozzle because of baking filament

rtyr, 2021-12-17:
It's already possible. You have to set up your tool change g-code.

Rough example:
[image]

n8bot, 2021-12-17:
Is there a way to shut down a tool after it is done being used in the print? That would be helpful. Otherwise it just hangs out at standby temp for a while.

bubnikv, 2021-12-20:
> Is there a way to shut down a tool after it is done being used in the
print? That would be helpful. Otherwise it just hangs out at standby temp
for a while.

No. That is a good idea to implement, for example for our XL.

AlexKid007, 2023-12-17:
This topic is from 2021, are there any Updates regarding this feature?

kubispe1, 2024-04-03:
SPE-2045

closed 2026-09-29 by prusaslicer-bot[bot]