prusa3d/PrusaSlicer #7517: Clicking "Seams" in dropdown doesn't select them

mpaw, 2021-12-18:
### Version
`Version 2.4.0-rc2+arm64`

Not specific to this version, I started using PrusaSlicer at 2.4.0 beta 2 and the issue has been there the whole time.

### Operating system type + version
`macOS Monterey Version 12.0`

### 3D printer brand / version + firmware version (if known)
Heavily modified Ender 3 Pro with custom Marlin firmware

### Behavior
Selecting `Seams` in the `Show` dropdown that is on the bottom left of the screen refreshes the dropdown but doesn't actually select `Seams`. I have to click `Seams` again and it will select on the second try. At this point you can select/deselect it and it works normally.

Video showing behavior
https://user-images.githubusercontent.com/5834637/146624658-f564abcf-301b-4740-885a-5f8f4e356401.mov

#### Project File (.3MF) where problem occurs
Not related to project files, always happens.

bubnikv, 2021-12-20:
This is a quirk on OSX only that we were not able to iron out. For PrusaSlicer 2.5 we most likely replace these controls with controls on the legend, thus the issue will go away.

lukasmatena, 2024-02-19:
We have replaced the controls with controls on the legend, thus the issue went away. Closing.