prusa3d/PrusaSlicer #7521: [Feature Request] prusa-slicer-console for Linux

503Dev, 2021-12-18:
### Version
2.3.3

### Operating system type + version
Ubuntu 20.x 

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
N/A
Feature Request

### Notes
When running Prusa Slic3r on Windows there is a command line tool, "prusa-slicer-console.exe" however there does not appear to be a Linux equivalent.  I have tried the ".AppImage" package and it forces a GUI version, in addition on a non-GUI environment it will not run and complains about missing GUI libraries.  Is it possible to provide an equivalent console/terminal for Linux environments? Thank you.

503Dev, 2021-12-18:
Please close this, it may be of use to make some notes for anyone in the future who needs this, it exists however the documentation to make it happen is not ideal.

First you must build prusa-slic3r yourself on Linux, if using a VPS you need 8GB of RAM to make this happen, it will crash with 4GB.

You can follow the build instructions here [I tested on Ubuntu 20.x]: https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

You will need to alter the cmake command to:
`cmake .. -DSLIC3R_STATIC=1 -DSLIC3R_GTK=3 -DSLIC3R_PCH=OFF -DSLIC3R_GUI=no -DCMAKE_PREFIX_PATH=$(pwd)/../deps/build/destdir/usr/local`

The key here is the _-DSLIC3R_GUI=no_ which will produce a console executable.

Once the build process is done navigate to the build/src directory and then you can execute the command line version with:
`./prusa-slicer`

You will be greeted with the familiar command line help text from "prusa-slicer-console.exe" in Windows.

bubnikv, 2021-12-20:
The prusa-slicer-console.exe does exactly the same as prusa-slicer.exe, only it opens a console.
On Linux there is no need for it, because the default prusa-slicer binary works with console by default.

I have updated the Linux how to build documentation:

> You need at least 8GB of RAM on your system. Linking on a 4GB RAM system will likely fail and you may need to limit the number of compiler processes with the '-j xxx' make or ninja parameter, where 'xxx' is the number of compiler processes launched if running on low RAM multi core system, for example on Raspberry PI.

bubnikv, 2021-12-20:
BTW thanks for compiling the no GUI version for us, we did not try for a long time :-)

closed 2021-12-20 by bubnikv

bubnikv, 2021-12-20:
I have updated the FAQ wiki
Frequently Asked Questions (FAQ) · prusa3d/PrusaSlicer Wiki (github.com)
<https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)#what-is-the-difference-in-prusa-slicerexe-and-prusa-slicer-consoleexe>

503Dev, 2021-12-22:
Awesome, thanks for the reply. I love prusa and appreciate you.  I have extensively tested the no GUI version now and have had great success.  It runs fine on far less than 8GB of RAM but you do need 8GB to compile.  Awesome work, thank you!

bubnikv, 2021-12-22:
Thanks. It is the G-code viewer that takes most of the RAM.

503Dev, 2021-12-22:
Agreed, I noticed the G-code viewer is a hog.  I also wrote my own G-code viewer from scratch a while ago for a job task -- using a WEBGL renderer and it was awful.  G-Code rendering is just messy, overall I think your version is satisfactory and I use it now.