prusa3d/PrusaSlicer #7525: PrusaSlicer 2.4.0-rc2 - Black labels

Shama71, 2021-12-18:
### Version
PrusaSlicer 2.4.0-rc2 - 

### Operating system type + version
Windows 11 x64

### 3D printer brand / version + firmware version (if known)
### Behavior
Run slicer in windows sandbox. 
[image]

#### Project File (.3MF) where problem occurs

bubnikv, 2021-12-20:
I am sorry, but I cannot reproduce. I tried Windows sandbox on Windows 11 for the first time. Did you set anything on the Windows sandbox configuration?

BTW it seems the sandbox does not pass through accelerated graphics, so PrusaSlicer falls back to software rendering, which makes it quite unusable. Thus we don't recommend sandboxing.

Shama71, 2021-12-20:
It looks like it's a sandbox issue
[image]
.

closed 2021-12-20 by Shama71