prusa3d/PrusaSlicer #7527: Crash after PC slept for several hours with PS open, then changed dark color mode

n8bot, 2021-12-18:
2.4.0-rc2 (n8 edition) win 10 64

So, this is probably a windows thing, or even something I did wrong, but I wanted to report it just in case it is helpful down the line.

I left PS open, and put my PC to sleep for several hours.

When it woke up hours later, everything was fine. I went to switch dark mode, and as soon as I accepted the preferences window, PS crashed.

[image]

```
Exception thrown at 0x00007FFBD6515F79 (vcruntime140.dll) in prusa-slicer-console.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.
```

Call stack:
```
 	vcruntime140.dll!00007ffbd6515f79()	Unknown
>	PrusaSlicer.dll!Slic3r::GUI::GUI_App::UpdateDarkUI(wxWindow * window, bool highlited, bool just_font) Line 1396	C++
 	PrusaSlicer.dll!Slic3r::GUI::GUI_App::UpdateAllStaticTextDarkUI(wxWindow * parent) Line 1479	C++
 	PrusaSlicer.dll!Slic3r::GUI::ConfigOptionsGroup::sys_color_changed() Line 755	C++
 	PrusaSlicer.dll!Slic3r::GUI::Page::sys_color_changed() Line 4125	C++
 	PrusaSlicer.dll!Slic3r::GUI::Tab::sys_color_changed() Line 1040	C++
 	PrusaSlicer.dll!Slic3r::GUI::MainFrame::on_sys_color_changed() Line 1049	C++
 	[Inline Frame] PrusaSlicer.dll!Slic3r::GUI::DPIAware<wxFrame>::force_color_changed() Line 182	C++
 	PrusaSlicer.dll!Slic3r::GUI::GUI_App::update_ui_from_settings() Line 1741	C++
 	PrusaSlicer.dll!Slic3r::GUI::PreferencesDialog::accept(wxEvent & __formal) Line 627	C++
 	[External Code]	
 	PrusaSlicer.dll!Slic3r::GUI::GUI_App::add_config_menu::__l2::<lambda>(wxEvent & event) Line 2269	C++
 	[External Code]	
 	PrusaSlicer.dll!Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams & params) Line 54	C++
 	PrusaSlicer.dll!Slic3r::CLI::run(int argc, char * * argv) Line 618	C++
 	PrusaSlicer.dll!slic3r_main(int argc, wchar_t * * argv) Line 838	C++
 	prusa-slicer-console.exe!wmain(int argc, wchar_t * * argv) Line 311	C++
 	[External Code]	
```

If the line numbers to match up to what you see (sorry!), look at this branch: https://github.com/n8bot/PrusaSlicer/tree/n8-240rc
I will not change that branch so the line numbers should match up there. I know... this is awkward... but I'm pretty sure nothing I changed should/would influence this. I make pretty small changes when possible.

bubnikv, 2021-12-20:
Three of us tried to reproduce, with no crash.

bubnikv, 2021-12-20:
I am leaving this issue open for some time for others to chime in.

n8bot, 2021-12-20:
Yeah this did not seem like something that would be easy to reproduce, but since I was debugging at the time I thought I would mention it.

lukasmatena, 2024-02-18:
I guess this is outdated now. Some possibly related problems were fixed since, it is likely that this was solved together with some of them. Closing.

closed 2024-02-18 by lukasmatena