prusa3d/PrusaSlicer #7529: Tangent lines are still around under PrusaSlicer 2.4.0-rc2

laptop-laptop, 2021-12-19:
### Version
2.4.0-rc2

_Use `About->About PrusaSlicer` for release versions_

2.4.0-rc2

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S+

### Behavior
* _Describe the problem_

I have been having issue with the kind of tangent lines mentioned in the below thread for over a year. Even with ironing and monotonic infill with 2.4.0-rc2, they are still there. Are they unavoidable due to the way slicers work? Is there any good way to get rid of them or make them less visible? 

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/anyone-know-what-could-cause-these-stray-tangent-lines/

ochm, 2021-12-19:
Do you use monotonous top infill?

laptop-laptop, 2021-12-19:
Yes

Top fill pattern: Monotonic
Bottom fill pattern: Monotonic
Fill density: 100%
Fill pattern: Rectilinear
Length of the infill anchor: 2.5 (set automatically)
Maximum length of the infill anchor: 12 (set automatically)
Flow rate: Tried 10-25%
Spacing between ironing passes: Tried 0.1 to 0.2mm

rtyr, 2021-12-19:
Please provide your 3mf project file.

laptop-laptop, 2021-12-19:
There is no specific project file. This is such a long standing issue that any surface with holes have it.

bubnikv, 2021-12-20:
@laptop-laptop Please provide a 3MF and photographs, otherwise we have nothing to work with and we will close your issue.

laptop-laptop, 2021-12-23:
Now it is not as bad as before. Thank you

closed 2021-12-23 by laptop-laptop