prusa3d/PrusaSlicer #7531: Crash after adding height range modifer

goldbird3, 2021-12-19:
### Version
PrusaSlicer-2.4.0-rc2+linux-x64-202112171415_dd0d1eacad42cfad815bd2da8847c390.AppImage

### Operating system type + version
NAME="Ubuntu"
VERSION="20.04.3 LTS (Focal Fossa)"

### 3D printer brand / version + firmware version (if known)
MK3S+ with MMUS2 Firmware is up to date

### Behavior
Slicer crashes

How to reproduce:
-open slicer in expert mode
-right click main window -> add Shape -> Box
-right click on green box object -> height range modifier
-In 'Height ranges' menu, input stop range to any number you like
-hit enter to finish
-cash

This only happens when I do it first in the session. If I do more stuff in the session before adding a heigt range it works with out a problem.

lukasmatena, 2021-12-20:
Reproduced on Ubuntu 16.04. But whatever the problem is, it is already happening in 2.3.1 (it works in 2.2.0).
ASAN output:
```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==20268==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x000000000000 bp 0x61b00011eee8 sp 0x7fff979de488 T0)
==20268==Hint: pc points to the zero page.
==20268==The signal is caused by a READ memory access.
==20268==Hint: address points to the zero page.

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (<unknown module>) 
==20268==ABORTING
```

enricoturri1966, 2021-12-20:
I cannot repro on Windows even using asan build.

bubnikv, 2022-06-29:
Fixed in PrusaSlicer 2.5.0-alpha2.
Closing.

closed 2022-06-29 by bubnikv