prusa3d/PrusaSlicer #7534: Support issue

gaminnick, 2021-12-19:
### Version
2.4.0-rc2_win64

### Operating system type + version
Windows 11
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
i3 MK3S

### Behavior
Slicing a part for Iron Man helmet the auto generated supports are putting supports where they are not needed.  When I try to block with paint on support it still generates the support.

#### Project File (.3MF) where problem occurs

[IronManHelmet.zip](https://github.com/prusa3d/PrusaSlicer/files/7741409/IronManHelmet.zip)

_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Snuff1eupagus, 2021-12-20:
You have support everywhere selected. As well, the supports would need to support down to the build surface in this situation even if you used paint on supports in that area, and had selected "For support enforces only" from the supports drop down. The angle of the helmet does not allow for full support from the lower area of the opening, supports drop straight down, and there is nothing to intercept them but the build surface. Support blockers do not work the way you think they do, it's not an error.

If you feel this area doesn't  need supports then use the paint on supports tool to paint them where you want them, or use the angle or smart fill feature to select or deselect the areas you want supported then use the "For support enforces only" selection from the supports drop down to see your selections honored.

This is not the place for lessons, you can find more information about this on the Prusa Forums you might find certain users there very knowledgeable.

Regards

gaminnick, 2021-12-20:
Thanks for the quick reply.  I thought you had to choose everywhere for support blockers to work as the other options were specific to other situations.  It seemed like blockers were pretty straight forward but I guess not.  I will look on the forums to learn more how this works.  The blockers did work on other areas, just not this odd support tower that is not really supporting anything.  Thanks again!!!

closed 2026-09-03 by gaminnick