prusa3d/PrusaSlicer #7538: carsh

ephramcohen, 2021-12-19:
### Version
_Version of PrusaSlicer used goes here_
Version 2.4.0-rc+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
windows 11

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
mk3s

### Behavior
* _Describe the problem_
I am creating a purge tester and setting all the purge amounts to 0.  Prusaslicer gives me an access violation when I slice my model.

* _Steps needed to reproduce the problem_
load model, slice.

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
model slices

* _Actual Results_
model doesn't slice and I get an access violation.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[purgeTest.zip](https://github.com/prusa3d/PrusaSlicer/files/7741801/purgeTest.zip)

bubnikv, 2021-12-20:
Your print is empty, because your models are too thin to produce any perimeter extrusions. If I enable "detect thin walls", your print will print just fine.

I suppose this issue is in PrusaSlicer 2.3.3 already. I will add a test for empty print, so PrusaSlicer will not crash but emit an explanation.

bubnikv, 2021-12-20:
Fixed with ef660873553c50a1c75feba617b16262d5e9edee

If there are no extrusions produced for printable objects, the G-code export is cancelled and the following notification is shown:

[image]

closed 2021-12-20 by bubnikv