prusa3d/PrusaSlicer #7541: Cosmetic change for the title of the 'Plater' tab in the English language version

twes100, 2021-12-20:
### **Version**

2.4.0-rc2 and older release versions.

### **Operating system type + version**

MacOS - but I assume it applies to other OS platforms as well.

### **Issue (Cosmetic)**

As a relatively new user of the English version of PrusaSlicer, my eye is always drawn to the rather incongruous title of the '**Plater**' tab. Strictly speaking, a 'plater' is someone (or something) that 'plates' - for example a 'metal plater'. I also wondered if the intended word was 'platter' but in English this more usually means 'large plate of food' - e.g. 'a platter of beef'.

I would like to propose to rename the tab to simply '**Print Plate**' which aligns with the English translation of the title used in the German language version - '**druckplatte**'.

Finally, PrusaSlicer is an amazing piece of software and kudos to all those who work so hard on the development and maintenance of it - much appreciated!

Danafra, 2021-12-20:
Since most of the community refers to it as "Print Bed", I'm not sure why it can't be named the same. Or maybe "Print Bed View".

twes100, 2021-12-20:
That would work just as well...

dartrax, 2021-12-20:
Also, native german speakers would normally use the term „Druckbett“ (Prind Bed), not „Druckplatte“ (Print Plate).

bbinnard, 2021-12-20:
The plater/platter etc. issue is an old one that has been discussed here before - at least once that I know of, and probably more times than that. As I recall, "plater" is intended to refer to the process of putting/printing something on the print plate (or bed), but "bedder" is not a word. 

My sense is that Plater has become somewhat of a trademark with this software and is unlikely to be changed anytime soon.

bubnikv, 2021-12-21:
I am not sure whether Plater was a typo of the original Slic3r author
Alessandro, however
Frequently Asked Questions (FAQ) · prusa3d/PrusaSlicer Wiki · GitHub
<https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)#what-is-the-correct-spelling-plater-or-platter>

dartrax, 2021-12-21:
Ahh yes, thanks for the link! So we already had that at in #2074 or #1226.

> My sense is that Plater has become somewhat of a trademark with this software and is unlikely to be changed anytime soon.

I have the same feeling but wasn‘t able to express that as accuratly as you 😉

n8bot, 2022-01-07:
We use the updater to update our system.

We use the charger to charge our batteries.

We use the helicopter to... helicopt our... cargo and passengers.

We use the plater to plate the parts on the print bed!

I hear it comes from *printmkaing*! A plater was someone who created printing plates to print pages for a book or whatever.

So, a plater is also a software function that serves that purpose! I computer used to be a person who performed the task of computing.

It's the tab that contains the software's plater function.

lukasmatena, 2024-02-19:
Closing. The meaningful discussion can still continue, but it will stay as it is.

closed 2024-02-19 by lukasmatena