prusa3d/PrusaSlicer #7542: Missing translation in dialog box

Alpacad, 2021-12-20:
### Version
2.4.0-rc2+win64

### Operating system type + version
Windows 11 Home 21H2

### Behavior
When toggling _Use system menu for application_ in _Dark mode_ section, on confirmation we are prompted to a dialog box whitch buttons aro not translated. My main language is italian, but this behavior is presenti in other languages too.
Attached screenshots are in italian and russian (where it is more obvious).

[image]

[image]

bubnikv, 2021-12-21:
This is a feature of wxWidgets that uses your system language for buttons. We will likely fix that in PrusaSlicer 2.5.

bubnikv, 2022-01-27:
Fixed in the upcoming PrusaSlicer 2.4.1-beta1.

closed 2022-01-27 by bubnikv