prusa3d/PrusaSlicer #7543: Can we choose which side PrusaSlicer uses to build the support for vertical holes?

laptop-laptop, 2021-12-21:
### Version
2.4.0-rc2

_Use `About->About PrusaSlicer` for release versions_
2.4.0-rc2 Win8

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 8

### 3D printer brand / version + firmware version (if known)
i3MK3S+

### Behavior
* _Describe the problem_

When printing a vertical wall with holes and chose Supports Everywhere, the slicer created some support materials around the holes. Is there a function to allow the user to choose which side of the vertical wall the slicer creates the support materials?

_Is this a new feature request?_

Yes if such function is not available yet.

Alpacad, 2021-12-21:
If I got it right, you mean **support blocker**?

You can either right-click on the model and select _Add support blocker_ and place it where you don't want the supports to be generated, or use the _Paint-on supports_ tool on the left toolbar, then paint the area with right-mouse button to exclude it from the support generator.

Note that this funcions are **available only** in _Advanced_ and _Expert_ mode.

laptop-laptop, 2021-12-22:
Thank you. I will try and report back if there is any problem.

closed 2026-09-29 by prusaslicer-bot[bot]