prusa3d/PrusaSlicer #7544: Can't add Filament in RC2

lfrancke, 2021-12-21:
### Version
2.4.0-rc2

### Operating system type + version

Linux / Arch 5.15

### 3D printer brand / version + firmware version (if known)

Prusa Mk3s+

### Behavior

When I want to add a new Filament I get the wizard to set up the printer instead and no way to select filament. The wizard also cannot be closed. I can only "Cancel" even after coming to the last page of the wizard.

[image]

lfrancke, 2021-12-21:
It seems as if the Filament menu does pop up only if I deselect my printer in the Prusa FFF menu and then immediately select it again.

lfrancke, 2021-12-21:
[image]

bubnikv, 2021-12-21:
Thanks for heads up, interestingly it was a Heisenbug and it happened very often on Linux, very rarely on other systems.
Fixed with 15d95e426496dbd4744722decb88ae40a05673b9

closed 2021-12-21 by bubnikv

lfrancke, 2021-12-21:
Thanks for the quick turnaround. I was doubting myself for a while.