prusa3d/PrusaSlicer #7545: GCode "G92 E0" does not always reset E steps to zero

YouMakeTech, 2021-12-21:
### Version
2.4.0-beta4+win64

### Operating system type + version
Windows 10 64 bits

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 Pro + Marlin 2.9.0.1

### Behavior
In the gcode generated by PrusaSlicer 2.4.0-beta4+win64, GCode "G92 E0" does not always reset E steps to zero. 
The following G1 command with extrusion extrudes a very large amount of filament.
This is not systematic. It can be seen at layer 25.7 mm or at layer 46.10 mm in the attached project file/gcode

[image]
[image]
[3mf Project File.zip](https://github.com/prusa3d/PrusaSlicer/files/7755034/3mf.Project.File.zip)
[gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/7755035/gcode.zip)

rtyr, 2021-12-21:
It is caused by misconfiguration in printer settings (combination of absolute distances for extrusion and custom G92 E0 command in "Before layer change g-code" script). 

Remove G92 E0 command from "Before layer change g-code", it will solve your issue.

[image]

Closing.

closed 2021-12-21 by rtyr

YouMakeTech, 2021-12-21:
Thanks for your reply. But I don't understand something. 
What is the point of using G92 E0 with relative extrusion? Relative extrusion already resets to zero by itself after each move. 
I think G92 E0 only makes sense with absolute extrusion.
Btw. I tested the exact project with the exact same settings with PrusaSlicer 2.4.0-rc2 and it works now... So the issue can effectively be closed now...

bubnikv, 2021-12-21:
> What is the point of using G92 E0 with relative extrusion?

Some firmwares keep an internal absolute extrusion counter, which loses
accuracy as it grows. Thus G92 E0 on every layer resets the firmware
internal counter and the accuracy is not being lost.

YouMakeTech, 2021-12-23:
Correction: I did some additional testing with version 2.4.0. The issue is not systematic but it still occurs from time to time.
Removing G92 E0 from the "Before layer change g-code" script or enabling relative extrusion distance in printer config solves the issue. But I think this is still a bug because "G92 E0" should reset the accumulated E-steps each and every time, whether I use absolute or relative extrusion distances.