prusa3d/PrusaSlicer #7546: Windows 10 v 21H2 breaks dark mode

n8bot, 2021-12-21:
Stock PrusaSlicer 2.4.0-rc2, dark mode title bars are now always white, as well as other elements incorrectly retaining light mode.

This seems directly related to installing 21H2, from 20H2 where everything worked fine.

n8bot, 2021-12-21:
@bubnikv I got you: https://github.com/notepad-plus-plus/notepad-plus-plus/issues/10810

n8bot, 2021-12-21:
prusadeps -> wx -> msw -> dark_mode -> dark_mode.h

bubnikv, 2021-12-21:
Oh well, what can we say, thanks Redmond. For us, at the time of 2.4.0 release it is an event of force majeure.

n8bot, 2021-12-21:
Lol I truly feel your pain. Luckily, this seems like an easy version check.

n8bot, 2021-12-21:
I snooped around NPP repo and see they also fixed some other problem due to windows 11: https://github.com/notepad-plus-plus/notepad-plus-plus/commit/5d086f93a80d275fcc5e4047ae49c614680a7e5b

There might be other recent additions. I suppose we/you should watch the NPP repo like a hawk for any dark mode issues, in case prusa users aren't catching or reporting them.

n8bot, 2021-12-21:
So I'm a newb when it comes to the deps, but I managed to jury-rig a patch (but I had to upload it to GH cause I don't know how to change the cmake script from using a DL .zip? lol. anyway, here's my patched wx if you wanna test:

https://github.com/n8bot/PrusaSlicer/releases/tag/wxpatchtest

The only difference is the version check numbers from `\wxWidgets-3.1.4-patched\include\wx\msw\dark_mode\dark_mode.h`

n8bot, 2022-01-06:
This is now fixed, along with those problems with app start on win10 home 21h2.

closed 2022-01-06 by n8bot