prusa3d/PrusaSlicer #7548: Gcode Preview showing wrong first layer temeprature

wmmc88, 2021-12-21:
### Version
_PrusaSlicer-2.4.0-rc2+win64_ (but also seems to affect earlier 2.4 betas)

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 11 Pro Insider Preview Build 22523.rs_prerelease.211210-1418

### 3D printer brand / version + firmware version (if known)
_FlashForge Creator 3 V2_

### Behavior
Prusaslicer is showing the wrong temperature for the first layer of a print. In the "other_layer_temp.3mf" file, it shows first layer as 245C, which corresponds to the "Other Layer" temperature instead of the "First Layer" Temperature. This file has the layer height and first layer height to be the same.

In "first layer height different.3mf", it shows the first layer temp as 0C which corresponds with the temperature of the 2nd (unused) extruder. In this file, the first layer height is different from the layer height.

In both scenarios, it shows the wrong layer temp. It should be 240C. It looks like the generated gcode still has the right temperatures  set, just not showing up properly in the gcode preview.

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
[first layer wrong temp.zip](https://github.com/prusa3d/PrusaSlicer/files/7756779/first.layer.wrong.temp.zip)

wmmc88, 2021-12-21:
[cylinder on prism_0.6n_0.2mm_L-PVA_R-PETG_36m_FFC3V2.zip](https://github.com/prusa3d/PrusaSlicer/files/7757015/cylinder.on.prism_0.6n_0.2mm_L-PVA_R-PETG_36m_FFC3V2.zip)

Another temperature issue. It looks like its rendering tools 2's temperatures for tool 1's move. then Tool 2's temperature is just showing as 0C. Once again, the actual gcode seems fine and has M104 commands for the correct temperatures in the correct places.

dartrax, 2021-12-22:
> Prusaslicer is showing the wrong temperature for the first layer of a print. In the "other_layer_temp.3mf" file, it shows first layer as 245C

I see 0C here in first layer, the same as in the other 3mf. However, it seems to be wrong anyway.

bubnikv, 2021-12-22:
Is it possible that you have the same issue as #7458?

bubnikv, 2021-12-22:
#7458 is indeed the same issue, see the following comment
IDEX temperature problem second extruder doesn't head up · Issue #7458 ·
prusa3d/PrusaSlicer · GitHub
<https://github.com/prusa3d/PrusaSlicer/issues/7458#issuecomment-991491246>

https://reprap.org/wiki/G-code#M104:_Set_Extruder_Temperature

*M104 in RepRapFirmware and Klipper*

*RepRapFirmware and some other firmwares support the optional T parameter
(as generated by slic3r) to specify which tool the command applies to.*

*However Marlin 2 documentation states, that the T parameter is supported
as well.*

*Set Hotend Temperature | Marlin Firmware (marlinfw.org)
<https://marlinfw.org/docs/gcode/M104.html>*

*Your block*

```
M104 S{first_layer_temperature[0]} T0 ; set right extruder temp
M104 S{first_layer_temperature[1]} T1 ; set left extruder temp
```

is not processed correctly by our G-code analyser. We likely need to
improve that.

For now, you can just emit M104 S without the T parameter and switch the
tools before setting the temperature.

wmmc88, 2021-12-22:
So the issue is just that prusaslicers gcoder viewer doesnt parse the M104 commands with the T parameter correctly?

bubnikv, 2021-12-22:
yes

boinappi, 2022-02-28:
Similar issue like https://github.com/prusa3d/PrusaSlicer/issues/7548#issuecomment-998965309
[image]

GCode seems fine and just a visualization issue.
PrusaSlicer 2.4.0

Garr-R, 2022-11-15:
Bump :) 

The GCode preview doesn't recognize the temperature of multiple extruders, only for Tool_0. The GCode is generated correctly! It just doesn't represent the temperature in the preview.

kubispe1, 2022-12-13:
In 2.6.0 G-Code preview is fixed and parsing G-Code M104 so:
- legend does not show 0 for the first layer.
- color for the layer's temperature is also separated and the legend should show the correct information.

closed 2022-12-13 by kubispe1