prusa3d/PrusaSlicer #7549: PrusaSlicer and 3D Connexion Wireless Space Mouse

plymatth, 2021-12-21:
### Version
2.3.3+

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S with MMU2

### Behavior
Installed PrusaSlicer on a brand new computer, along with 3D Connexion Wireless Space Mouse and Fusion 360.  Space Mouse works as expected with Fusion360, only moves 1 axis in PrusaSlicer.  Older computer running same setup works fine.

* _Steps needed to reproduce the problem_
Reinstalled PrusaSlicer a few times.  Sometimes I can get the plater to spin along the horizontal access.  After a reinstall, it once was able to zoom in/out (but no rotation).  I have also tried with and without the USB cord connected, no impact.  I've also checked "Enable Support for Legacy 3D Connexion devices"

bubnikv, 2021-12-22:
Please check
https://www.help.prusa3d.com/en/article/3dconnexion-devices_1511
and the comments below.

The 3DConnexion drivers are supposed to install configuration file specific
for PrusaSlicer:
https://github.com/prusa3d/PrusaSlicer/issues/4104#issuecomment-707727500

I see in the comments below
https://www.help.prusa3d.com/en/article/3dconnexion-devices_1511
that maybe the newest 3DConnexion drivers don't install C:\Program
Files\3Dconnexion\3DxWare\3DxWinCore\Cfg\prusa-slicer.xml
Would you please confirm that? If it is the case, then we have to notify
3DConnexion to fix their drivers.

plymatth, 2021-12-22:
Solved!   Well, kinda.  I removed the prusaslicer.xml driver from the 3D Connexion folder, and everything works.

plymatth, 2021-12-22:
Solved by removing prusaslicer.xml

closed 2021-12-22 by plymatth

szafran81, 2021-12-23:
Worked for me (don't forget to turn off the app while removing the file and then start it back again).

enricoturri1966, 2022-01-06:
Driver: 3DxWare64_v10-8-4_r3407.exe

I was able to get my SpaceMouse Pro Wireless working with PrusaSlicer by editing the file C:\Program Files\3Dconnexion\3DxWare\3DxWinCore\Cfg\prusa-slicer.xml
using C:\Program Files\3Dconnexion\3DxWare\3DxWinCore\Cfg\Blender.xml as reference.

This is the edited file:

```
<?xml version="1.0" encoding="utf-8"?>
<!--
(c) Copyright 3Dconnexion. All rights reserved. 
$Revision: 16358 $
-->
<AppCfg  xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" Default="true" CfgFormatVersion="1.3" ThisFileVersion="1.0">
  <CfgProperties>
    <ID>ID_PrusaSlicer</ID>
    <Name>PrusaSlicer</Name>
    <InheritsFromID>ID_Default_Cfg</InheritsFromID>
  </CfgProperties>
  <AppInfo>
    <Signature>
      <Name>PrusaSlicer</Name>
      <ExecutableName>prusa-slicer.exe</ExecutableName>
      <Transport>RawInput</Transport>
    </Signature>
  </AppInfo>
  <AuthorInfo>
    <DriverVersion>17.6.3.16588</DriverVersion>
    <Created>
      <Tool>3DxService.exe</Tool>
      <Version>17.6.3.16588</Version>
      <Date>2020:4:3:1:10:16:46:953</Date>
      <User>prusa</User>
    </Created>
    <LastChange>
      <Tool>3DxSmartUI.exe</Tool>
      <Version>1.2.19261.16572</Version>
      <Date>4/1/2020 10:22:47 AM UTC</Date>
      <User>prusa</User>
    </LastChange>
  </AuthorInfo>
</AppCfg>
```

bubnikv, 2022-01-06:
and we have reported to 3DConnexion

n8bot, 2022-02-04:
Since the xml lists "prusa-slicer.exe" does the space mouse work correctly when running the gcode viewer or the -console.exe ?

RALaRock, 2022-02-12:
Editing prusa-slicer.xml worked for me as described by [enricoturri1966]. Just a bit more info. In case it helps. The change required is in this line:
    <InheritsFromID>ID_Default_KMJ_Cfg</InheritsFromID>

changing it to this (removing "_KMJ"):
    <InheritsFromID>ID_Default_Cfg</InheritsFromID>

I closed Prusa Slicer. Made the edit and saved it. Then opened Prusa Slicer and it works fine. Only tested this on version 2.4.1-beta1.

plymatth, 2022-02-12:
Updated 3D Connexion drivers to version 10.8.5.3424.  This worked for me with no issues for Space Mouse Wireless