prusa3d/PrusaSlicer #7550: No error message in UI for empty files

thijstriemstra, 2021-12-22:
### Version
2.4.0 rc2

### Operating system type + version
Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Prusa mini+ 4.3.4

### Behavior
When importing an empty STL file I see no error message in the UI, only in the terminal Prusa Slicr was launched from:

```
[2021-12-22 01:59:34.747014] [0x00007fd52a8b08c0] [error]   stl_open_count_facets: The input is an empty file: /home/foo/projects/bar/cad/audio-jack-cap.stl
```

bubnikv, 2021-12-22:
I see the following error message in UI if trying to open an empty file.

Tested on Linux.
I created an empty STL from terminal:
$ touch empty.stl

then I opened the empty STL and I got the following error message:
[image: image.png]

thijstriemstra, 2021-12-22:
thanks for the feedback @bubnikv. Unfortunately the image doesn't show up in github ui:

```
[image: image.png]
```

bubnikv, 2021-12-23:
Attaching images from e-mail to github seem to become broken sigh.

thijstriemstra, 2021-12-23:
I see the same error message when importing an invalid STL, but **not** when importing a valid STL, then reloading the updated (incorrect) STL on the build plate.

bubnikv, 2022-01-27:
> I see the same error message when importing an invalid STL, but not when importing a valid STL, then reloading the updated (incorrect) STL on the build plate.

Fixed in upcoming PrusaSlicer 2.4.1-beta1.

closed 2022-01-27 by bubnikv