prusa3d/PrusaSlicer #7554: Gcode extractor preceeding 0 not in gcode anymore New Slicer 2.4.0

kevinmichaeldudley, 2021-12-22:
### Version
_Version of PrusaSlicer used goes here_
2.4.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Mk3s+

### Behavior
* _Describe the problem_
* Viewing Gcode preceeding extruder "0" now stripped.
When I am tracking "extrusion issues" I often am reading the gcode to look for errors etc..
My visual scanning is now "slowwed" down and potential mis-reads due to alignment issues.
The new "0" stripped format makes it more difficult to follow the extruder gcode outputs.
I know it saves "space" but it is VERY LITTLE
FYI: I have a print farm with 200+ products so I get deep into making code better or fixing "quirks".

* New code 2.4.0 slicer follows...
 G1 X163.956 Y100.423 E1.18963
 G1 X164.209 Y101.486 E.02846       <-------- Missing "0"

Old Code 2.3.0 Slicer follows...
  G1 X163.956 Y100.423 E1.18963
  G1 X164.209 Y101.486 E0.02846      <----------- "0" included much EASIER to quickly read.

Maybe in configuration I can turn this back ON?????

Thanks!!

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

neophyl, 2021-12-22:
Duplicate of already closed issue https://github.com/prusa3d/PrusaSlicer/issues/7505

bubnikv, 2022-03-25:
See https://github.com/prusa3d/PrusaSlicer/issues/6996#issuecomment-1023408827
Closing

closed 2022-03-25 by bubnikv