prusa3d/PrusaSlicer #7555: Import config from gcode not working and shows error message

PingPongLee, 2021-12-22:
### Version
2.4.0 stable

### Operating system type + version
Windows 10 21H2 (Build 19044.1415)

### 3D printer brand / version + firmware version (if known)
MK3S+ running firmware 3.10.1

### Behavior
Using the configuration import through File -> Import -> Import config, to import a print profile from a gcode file created in PrusaSlicer for the same printer, in an earlier version (2.2+) i get the following error message:

"Suspiciously low number of configuration values extracted from C:\Users\Daniel\Downloads\Lava Lamp testing_0.3mm_ASA_MK3S_3h25m.gcode: 0"

I've tried with another gcode file in which I want to import the same print profile, but the same error appears. I've attached the gcode file in question

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[Lava Lamp testing_0.3mm_ASA_MK3S_3h25m.zip](https://github.com/prusa3d/PrusaSlicer/files/7762576/Lava.Lamp.testing_0.3mm_ASA_MK3S_3h25m.zip)

PingPongLee, 2021-12-22:
That line seems to be from a plugin in octoprint, which I do indeed use. 

You would think that the parser would just ignore it though

bubnikv, 2021-12-22:
Your scenario never worked in PrusaSlicer. You generated your G-code
with PrusaSlicer 2.3.1, even PrusaSlicer 2.3.1 cannot import your G-code.

> You would think that the parser would just ignore it though

Yes, it is not ideal.

Your issue seems to be a duplicate of #6730. We improved already, thus the
new G-codes will be handled properly.

From the changelog of PrusaSlicer 2.4.0-alpha1:
Release PrusaSlicer 2.4.0-alpha1 · prusa3d/PrusaSlicer · GitHub
<https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1>

   - Configuration stored into G-code is newly delimited with ";
   prusaslicer_config = begin" and "; prusaslicer_config = end". These
   delimiters look like any other key / value configuration pairs on purpose
   to be compatible with older PrusaSlicer versions trying to parse
   configuration from G-code. The G-code viewer reads the "; generated by
   PrusaSlicer ...." header and if the G-code was generated by PrusaSlicer
   2.4.0-alpha0 and newer, it expects that the configuration block is enclosed
   by the new delimiters. This makes the G-code parser more robust going
   forward and G-codes can now be loaded by the G-code viewer even if an
   OctoPrint plugin adds its own comments at the end of the G-code after
   PrusaSlicer's configuration
   #6730 <https://github.com/prusa3d/PrusaSlicer/issues/6730>.

bubnikv, 2022-06-29:
Implemented with PrusaSlicer 2.4.0-alpha1.
Closing.

closed 2022-06-29 by bubnikv