prusa3d/PrusaSlicer #7556: Voron Trident Z incorrect

mtaumike, 2021-12-22:
### Version
Release 2.4

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
Voron Trident and all build versions by spec are 250mm on the Z

All nozzle sizes show the same 230mm instead of 250mm.

### Behavior
Voron Trident printer preset is set to 230mm and should be 250mm.

#### Project File (.3MF) where problem occurs
N/A

rtyr, 2021-12-22:
Sorry I don't understand, we don't provide any Voron Trident printer preset.

mtaumike, 2021-12-22:
> Sorry I don't understand, we don't provide any Voron Trident printer preset.

Hmm.. I'll wipe out my install completely (I have backups) and launch 2.4 again from scratch. Could be some 'leakage' from futzing with SuperSlicer as well and hopping systems.

Thanks for the response. I'll dig further to make sure it is an user error of my own.

closed 2021-12-22 by mtaumike

rtyr, 2021-12-22:
SuperSlicer also doesn't include preset for Voron Trident, so it will be probably some imported profile.

bubnikv, 2021-12-22:
Everyone is most welcome to offer new printer profiles.
However PrusaSlicer cannot currently download printer thumbnails for the
wizard, thus adding a new printer over the air is not possible yet.