prusa3d/PrusaSlicer #7559: Prusaslicer installs as V2.3.3

jokohoko, 2021-12-22:
### Version
V2.4.0

### Operating system type + version
Windows 10
Installed using the new experimental installer 

### Behavior
I just installed the new V2.4.0 using the experimental installer. And made sure to remove other version beforehand. But in my programs list it now shows as V2.3.3. Once i open it the title of the software does show as V2.4.0.
Thus i think the link's name has not been updated?
Not a major issue, just confusing a bit since i don't have V2.3.3 installed on my system.
[image]

rtyr, 2021-12-23:
@Janys FYI

lukasmatena, 2024-02-19:
Outdated, closing.

closed 2024-02-19 by lukasmatena