prusa3d/PrusaSlicer #7561: MMU2S change the extruder print order

Laky1991, 2021-12-22:
### Version
2.4.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3S+ FIRMWARE 3.10.1  MMU2S FIRMWARE 3.10.1 (1.0.6)

### Behavior
I cannot set printing in an extruder order other than 1-5.
The 1st extruder and then the 2nd extruder are always printed.
I need to change this order.
I want to start printing from extruder 2 and then continue printing with extruder 1.
This way I always have to manually change the filaments in the MMU2S stack, because it can't be changed in the prusaslicer
For example, in the printing process business card:
1. extruder black color and 2. extruder white color
Black business card and white text.
 The business card is printed first and then the text
I would like the first text to be printed and then the business card.

This is a new feature request.

Darkknight33, 2021-12-22:
I'm not a dev, but can you help me understand why it's important the text be printed first? The entire print is done by layer, so regardless of A->B or B->A the final result should look the same.

Laky1991, 2021-12-22:
When you print a label, the colors change in one layer. 
The view side lies on the bed.
When is printed first business card , there is not enough space inside the perimeter for the text. 
It looks better when printed the other way around.

Darkknight33, 2021-12-22:
Have you tried the external perimeters first option? The output on the printer should always resemble what's actually in the slicer. 

If the text part of the print isn't fitting when you go to actually print, it sounds like you may need to calibrate your extrusion width and/or elephant's foot compensation. If your first layer is too close to the bed, it may also be overly smooshing the text.  Simply raising up your Z a little bit might solve this issue.

Lastly, you could also scale up the XY for business card (but not the text) slightly, like say 0.25% and get more room for the text to print.

Laky1991, 2021-12-22:
I've tried different settings, but it's best to print the first text.

Darkknight33, 2021-12-29:
Why don't you post your 3MF and I let me take a look at it?

Laky1991, 2021-12-31:
[Logo multimaterial.zip](https://github.com/prusa3d/PrusaSlicer/files/7795449/Logo.multimaterial.zip)

Darkknight33, 2022-01-01:
This was printed from your 3MF. I used the PLA I had already in there from a previous print, and so my filament profile along with it. I also did not use whatever changes you made to your start/stop Gcode. The text printed first on it's own. It's not terrible looking, and I wouldn't say the text looks 'squished'. I'd change a few things to get it to print smoother, but this was your file so I wanted to change as little as possible. I don't know exactly what problem you are experiencing, but this can absolutely be tweaked to be smooth in just your print settings. Just lowering the extrusion multiplier or elephant foot compensation would probably push enough filament in to make this smooth on my print. I also think Hilbert Curve bottom layer is a bad choice. It exacerbates the under/over extrusion problems. If you want it to look smooth, monotonic, or if stylized then maybe archimedian chords. 

Personally, I'd print this face up with the text sitting _above_ the plane of the card. It's for show, so print the card in 3 dimensions. Nobody is actually gonna keep it in their wallet, so it doesn't need to be super practical.  It also neatly avoids whatever problem you are having. 

[image]

martijnvankekem, 2022-01-25:
@Darkknight33 that you were able to optimize to printability of this specific STL, doesn't render this feature request redundant. There are a lot of practical reasons where changing the order of the extruders makes a difference, even if this STL came out nice using the 1-5 method.

Darkknight33, 2022-01-25:
> @Darkknight33 that you were able to optimize the printability of this specific STL, doesn't render this feature request redundant. There are a lot of practical reasons where changing the order of the extruders makes a difference, even if this STL came out nice using the 1-5 method.

@martijnvankekem Did I say the feature request was redundant someplace? Rereading my replies to @Laky1991, and I don't see that. I was trying to help him get his print fixed now, rather than waiting months for a feature that may never be implemented and probably won't actually solve the problem he's having. I don't believe I've indicated anywhere that this feature request was useless or should not be implemented.

Since you brought it up though, what is the practical benefit of having software control over the extruder print order when you can literally just load the filaments into the mmu2s in the order you want them to print? Is it simply to save you from having to swap filaments around before starting a print?

Regardless, if this is something you really want to implement, by all means go ahead. I'm not trying to stop you.

MysticGringo, 2023-02-10:
I know this is an old topic... but I thought I'd second the idea of being able to decide which filament the printer chooses first. Similar issue as the OP, I am printing labels for things on my boat, some that go onto the control panel, and some that go with N2K electronics T's.

I keep white in position 1, and black in 2 on my MMU, and sometimes I want to print white text on black, others black text on white. The labels are using pretty small text as that's the space I have, and it just works better having the text printed first, and then the background second.

I am sure through tuning the printer even more that I can get better results, but I've had the best results printing text first. Yes, I could swap the filaments, but that's a bit of a pain to do for each print if you want to change the order, especially if the print would respect the order the parts are in the print list.

mike-serinus, 2023-07-31:
I second this issue. It is a major problem, particularly with narrow objects such as text on the first layer. If the text is not printed first, the surrounding color tends to spread into the text zone, making the text thinner than it should be. Compensating with line width or extrusion multipliers does not help, as it affects other aspects.

strud, 2023-08-08:
I would very much like this feature also, for the same reason ie text inlays

janakelarsson, 2024-01-17:
I agree. I am now going to unload and load my MMU3 with the same filaments in a different order

AndrewFarley, 2024-06-10:
I came here to make this feature request also, seems someone beat me to it.  I have a great STL and picture capturing just why this feature is so simple but important.

I had to do som trickery to convince my slicer to print the white (details)  before the black.  I did this by having a second object that uses a lot of that color.  It seems to choose the first extruder to use being the one that uses the most filament.  But, I can't always convince the slicer to do it consistently.

If I could simply choose which extruder to start with (on at least the first layer, I don't personally care about the other layers) then it would really make beautiful first-layers.  Please see image.  I bet you can tell which one printed the black first, versus the white first.

[image]

mj8w, 2024-08-08:
I too would really appreciate this. In my case, I prefer the background printed first in some cases, but my text is a bit larger that some that I am seeing in others' work.

I often print something, then figure out that it should have been the other way around, and then go through the effort to unspool and then rethread everything again. It is a hassle that should be a simple software thing.

HonzaJaros, 2024-08-18:
same the last layer. just printed black text n white, if the black text is first it would not stain the white. I woish there will be select box selecting what extruder you want to finish o start th layer. or selectable sequence

jalicz, 2025-01-08:
I join the request, I would very much welcome this feature

closed 2026-09-29 by prusaslicer-bot[bot]