prusa3d/PrusaSlicer #7562: Bug in the new gcode viewer

MikeSavad, 2021-12-22:
This is a feature i've been wanting... Anyway, when I view a gcode, it will show the model, but not the supports. It will show the head filling in supports, but the model is otherwise hollow.  (Unless its a setting, but I can't see why it wouldn't show it)

This is only true of the show the gcode, not the preview slicer.

Also on the topic of that, can the preview show me what the gcode see's? I paint a line on the side to purge and I would like to see what the head does there as well. Otherwise I would have to save it, then view it. 

----Mike Savad

### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2021-12-22:
Please attach data: 3MF, gcode or both, possible with screenshots.
You will need to zip 3MF, github does not allow attaching 3MF files
directly.

MikeSavad, 2021-12-22:
That's weird, now its showing it. I wonder what happened?  Oh well never mind.

----Mike Savad

closed 2021-12-22 by MikeSavad