prusa3d/PrusaSlicer #7563: Feature Request: Object dimension extension between point A and B

fuirippudotca, 2021-12-22:
problem 
changing "Z height" dimension of the model in PrusaSlicer ( or any Slicer ) would make the holes and indent for the nuts "oval/long"

Solution
The requested feature would extend a portion of an object between a user selected Point A and Point B so a user can extend the object dimentions overall without changing the dimensions for "fitted" objects such as hole for a screw or indent for a bolt. 

Example Model
[image]

Point A example
[image]
Point B example 
[image]

bubnikv, 2021-12-22:
We will work on a measurement tool. It is then a matter of basic
arithmetics to achieve what you need.

fuirippudotca, 2021-12-22:
Interesting, I hope this is what I'm looking for.

dartrax, 2021-12-22:
@bubnikv, looking at the Title I also imagined some kind of measuring between two points, but what the author requested is actually a tool to modify the STL.
fuirippudotca wants to split a model at a given Z plane, move both parts a distance apart, and connect the cutting surfaces by extruding them in Z direction.

fuirippudotca, 2021-12-22:
@dartrax That is correct

bubnikv, 2021-12-23:
@dartrax Thanks for explaining, sometimes it takes quite some mental effort to understand the request and I have to skim through a lot of communication and yet to do some work in addition.

So the feature is about being able to shrink and expand an object (for example a box) by extruding along a plane or by removing a section.

neophyl, 2021-12-23:
A task way more suited to some proper CAD software.  This process would take about 30 seconds to do in Blender once the model was loaded.  I only use Blender as an example as I am more familiar with that software but Its probably just as easy to do in other CAD packages.

You can actually do this already in PS but its a pain in the rear.  You load 3 instances of the model.  You cut them to leave the bottom with the holes, the top with the holes and the middle section with no holes.  You then stretch the middle in Z to give the required extra length then merge them and stack them on top of each other.  Basically a time consuming task as a slicer isn't and shouldn't really be used to do CAD.

julie777, 2023-01-27:
I favor enhancing the cut tool.
1. select a vertical or horizontal plane
2. move the plane to the desired location
    a. cut off the top or bottom
    b. insert a specified length of the cross section
    b. remove a specified length and direction from the cross section
(This is similar to push/pull in Fusion 360)

edwardhorsford, 2023-06-11:
I'd love the ability to stretch or trim part in a single dimension at a chosen point. I frequently download models which are otherwise perfect, but for my usage should could be narrower or wider.

I could imagine it working very similarly to the newly improved cut tool. For extending, you'd place the plane where you want it and either drag or enter a numeric extension amount. For trim, you could have two parallel planes, or else numeric input for their separation.

You can almost do trimming now with multiple cuts - but it's super frustrating as PrusaSlicer insists on moving parts after cutting them - so different cut parts no longer line up.

KlausKraemer, 2025-08-17:
Could "Resizing / rescaling multipart models feature" https://github.com/prusa3d/PrusaSlicer/issues/14537 be moved here? Thanks!

fuirippudotca, 2025-10-07:
Thank you for the GIF it demos exactly what I wished to see.

KlausKraemer, 2025-10-07:
Wow, seems great!

Would this also work with models coming ready divided in  multiple parts like I mentioned above?

alidorauxier, 2025-11-17:
This would be a really good feature! It's needed surprisingly often for modifying things like cases and brackets.
It occurs to me though that this only works where it's just the two ends that need to remain 'unscaled' - objects that have multiple elements (like holes or tabs) along the model would still get distorted/stretched.
Maybe we could support using multiple slices concurrently? This would scale each segment equidistantly.
To be fair, the same could be achieved just by measuring, calculating and reapplying the extend/shrink for each safe zone.

Daniel599, 2025-11-17:
@alidorauxier 
My current implementation moves all the parts around the selection, so if you choose 2 parts above each other to scale, sadly it won't work as you describe, I think you can do it in two phases.
I did provide info about the added length when you drag the mouse to scale.

superDIYer, 2026-01-06:
Really can't wait for this features. Despite how flipping easy its claimed to be in Blender, I'll be dipped if I know how.

Daniel599, 2026-09-18:
Anyone who uses prusaslicer 3.0 can tell me if I should try to re-add this feature? Or is there something like that already?
Thanks

KlausKraemer, 2026-09-18:
I see still too many issues with PS 3.x popping up. That is, why I am reluctant to try the alphas. If no one can tell you, whether it's implemented, I ask you for waiting until you know and adding the feature, if it's missing.

Thanks!!!