prusa3d/PrusaSlicer #7564: Button in focus is not highlighted anymore in PS 2.4

Le0Michine, 2021-12-22:
### Version
2.4.0+win64

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Previously in PS v2.3 when I opened "send G-code" dialog (ctrl+shift+G) I could hit enter to send the file to my printer and start printing job given that "start on upload" checkbox was selected. In the new PS there is no checkbox anymore, instead there are three buttons: "Upload", "Upload and Print", "Cancel". By default if I hit enter "Upload" button will be clicked. I can press tab once to shift focus to "Upload and Print" button and then press enter to sent g-code and start printing, however there are couple issues with that

- it adds an extra action every time I send something to print which is noticeable regression comparing to the experience with previous version. 
- the button in focus is not highlighted in any way, before there was a visible outline allowing to confirm which button will be pressed on enter

* _Describe the problem_
* in PS with any sliced model hit Ctrl+Shift+G
* "send G-code" dialog appears with "Upload" button highlighted and "Start on upload" checkbox present
* _Actual Results_
*  "send G-code" dialog appears with no button highlighted, and no "Start on upload" checkbox available
  * _Screenshots from __*PrusaSlicer*__ TBD

_Is this a new feature request?_
kinda, more like restoring original behavior request

#### Project File (.3MF) where problem occurs
N/A

n8bot, 2021-12-22:
Good find. I found a fix.

If you can compile yourself you can get the fix by cherry-picking this commit: https://github.com/n8bot/PrusaSlicer/commit/8102e904e33a08a0e3678ac465346bf88b6c63ff

With this change, the buttons will highlight like they do when you hover them with the mouse, when they are selected with the keyboard.

Otherwise, I'm sure it will be fixed in the next release.

Le0Michine, 2021-12-23:
Thanks! Appreciate the quick response and action!

Though it's a bit a bummer that the checkbox is going away

n8bot, 2021-12-23:
Which checkbox do you meant? Do you mean the border on the buttons?

n8bot, 2021-12-23:
Oh, I see I didn't notice your comment about that. I do not know about that. I'm not part of the prusa team just a devout fan. Lol. Perhaps the functionality has moved around a bit. I know that they definitely added/worked on support for direct uploading to certain printers.

bubnikv, 2021-12-23:
@Le0Michine

Thanks for the report. We wish more people would test the betas, so we would get these issues reported before the final release.

n8bot, 2021-12-23:
I test the betas! :)

I realize that my fix breaks down when the user passes the mouse past the button -- this "steals" the highlighting away from the focus event. Instead, perhaps, there could be two bools in the lambda: one for hover, one for focus. The focus would be for actual focus, and the hover would be for what is now called focus, but is hover-highlighting. Then, importantly, the visual change would have to be different than the one performed by hover, but not mutually exclusive to the hover. So both states can be triggered together, or disabled separately.

MikePfaff, 2021-12-24:
Please note that hitting  "tab" to switch the focused button from "Upload" to "Upload and Print" in the dialog doesn't work at all on macOS. The only way to trigger "Upload and Print" on macOS is to click on the button with the mouse, which is rather tedious compared to "Just hit enter" (if you have in the past at some point checked the checkbox) in previous versions.

YuSanka, 2022-01-05:
> Please note that hitting "tab" to switch the focused button from "Upload" to "Upload and Print" in the dialog doesn't work at all on macOS. The only way to trigger "Upload and Print" on macOS is to [...]

And that is right. To change this behaior you have to change your System Settings: 
`System Preferences -> Keyboard -> Shortcuts -> Check "Use keyboard navigation..." at the bottom.`

Note, that you can to use next Keyboard Shortcuts :
**⇥ (TAB)** will move between buttons.
**esc (ESC)** is cancel.
**space (SPACE)** selects the active button (blue, outline).
**↩ (RETURN)** is OK or the default button (blue, pulsing, filled).

MikePfaff, 2022-01-15:
@YuSanka That can have effects on other applications, which is not desired. My suggestion would be to bring back the checkbox that has worked nicely for a long time.

ChrisHeerschap, 2022-02-01:
Mac OS users should not be forced to change our system settings because of a bad decision in taking away our ability to choose a default action. +1 on @MikePfaff 's request to return to the ability to select a default action with a checkbox.

closed 2026-09-29 by prusaslicer-bot[bot]