prusa3d/PrusaSlicer #7565: Retraction and Z-stroke on the supports

Laky1991, 2021-12-22:
### Version
2.4.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3S+ FIRMWARE 3.10.1 MMU2S FIRMWARE 3.10.1 (1.0.6)

### Behavior
Retraction and Z-stroke on the supports
Retraction and Z-stroke are hardly performed in the first layer.
In the previous version of the prusaslicer, retractions were performed on small shapes in the first layer of supports. In the new version, retractions are not performed and the thread is stretched over the bed.
In the enclosed 3MF, retraction is not performed on small rectangles on the bed at the first layer, and at rapid traverse, the entire rectangle is sometimes torn off because retraction and lifting are not performed.

#### Project File (.3MF) where problem occurs

[Podložka.zip](https://github.com/prusa3d/PrusaSlicer/files/7764487/Podlozka.zip)

bubnikv, 2021-12-23:
I think your issue is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7353. Is it so?

Laky1991, 2021-12-23:
yes i didn't notice that.

rtyr, 2022-01-03:
Closing as a duplicate.

closed 2022-01-03 by rtyr