prusa3d/PrusaSlicer #7568: Add object on platter does not follow symbolic links

rjwgnr27, 2021-12-22:
### Version
Version: 2.4.0
Build: PrusaSlicer-2.4.0+flathub.org

### Operating system type + version
Operating System: Unix
System Architecture: 64 bit
System Version: Linux 5.15.10-100.fc34.x86_64 x86_64
Total RAM size [MB]: 16,752MB

### 3D printer brand / version + firmware version (if known)
Anet A8 

### Behavior
I have just upgraded to 2.4.0 from 2.3.?, where the following worked: I have a directory on a remote share, with a symbolic link to that directory in my home directory. Trying to add an object from the "plater" screen, the open file dialog shows my home directory, and the physical directories within; however, I do not see the link to my shared directory.

* Problem: On the "Plater" screen, when adding an object, the open file dialog does not show or follow symbolic links
* Steps: Create a folder somewhere. In home directory create a symbolic link to that folder. Open PrusaSlicer to "Plater" screen. Click the "Add" button, and look in your home directory. The link to the linked folder is not visible.

* Expected: To see the link to my shared folder
* Actual: Only physical directories, and object files (*.stl, etc) are visible. The symbolic link to the shared directory is not visible. Note, the symlink name "shared", so not a "hidden" file.

The above worked in 2.2.* and 2.3.*, so appears to be a regression in 2.4.0.

#### Project File (.3MF) where problem occurs
There is no project file associated with this.

bubnikv, 2021-12-23:
Please test with our AppImage. If everything works with AppImage and you
have the issue with FlatPack distribution, then please consult with the
FlatPack maintainer. If the FlatPack maintainer finds out the problem is on
our side, he will contact us.
@xarbit

xarbit, 2021-12-23:
@rjwgnr27

xarbit, 2021-12-23:
@rjwgnr27 thats interesting. please let me know if this behavior only happens when using the flatpak and I'll have a look at it.

rjwgnr27, 2021-12-23:
Hi Jason,

I did try the appomage version, and that works as expected. So this does
appear to be a flatpak issue.

Thank you,

  --rick

xarbit, 2021-12-24:
Ok, thanks for the feedback I’ll look at it.

closed 2026-09-29 by prusaslicer-bot[bot]