prusa3d/PrusaSlicer #7570: New feat req : Storage of the linear advance coefficient

smoosun, 2021-12-23:
### Version
All versions (currently 2.4.0)

### Operating system type + version
All

### 3D printer brand / version + firmware version (if known)
PRUSA printers supporting linear advance control (at least)

### Behavior
It would be nice to be able to store the linear advance value (dependent on each filament) in the PrusaSlicer filament profiles, just like the extrusion coefficient.

If not entered, then default value of the machine.
If set, then the change is reflected in the produced G-Code.

This is a request for new functionality.

Thank you !

n8bot, 2021-12-23:
Can't you do this by adding it as a custom start g-code for the filament?

dartrax, 2021-12-23:
duplicate of and answer in #5138, although that is already closed.
I‘d like to add that I use LA not for infill, so different LA factors for different feature types would be great. I solved that in Simplify3D with the integrated post processor. Not sure if I have that already in my PS profiles or if it‘s possible…

closed 2021-12-23 by smoosun