prusa3d/PrusaSlicer #7573: Not Merging Raft, Creating double paths

VexMex, 2021-12-23:
### Version
2.4.0+win64

### Operating system type + version
Windows 10 20H1

### Behavior
Prusa does not Combine rafts and creates gcode that goes twice over a printed surface

Put booth gears clsoe to each other, let them slice with rafts

Expected Result:One Big raft area that combines the two single rafts without creating double paths

Actual Results: Does not Create a single merged raft and instead moves the two single rafts over each other, creating paths that go twice over a printed area

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_
No, Its a bug

#### Project File (.3MF) where problem occurs

[Wheel 2.zip](https://github.com/prusa3d/PrusaSlicer/files/7768411/Wheel.2.zip)

neophyl, 2021-12-23:
This is not a new issue.  Its been reported many times before (1275, 4517 but theres probably others too).   Its always done that as it slices each object individually.  You can overlap different objects to and it will do the same with normal perimeters and infill etc.

viper0078, 2021-12-23:
I have stated the same thing before. It takes a bit of work, but you can use the merge function to avoid this problem.
Select multiple objects, and then select "merge" from the right-click menu.
[image]

The merged objects will have their rafts merged.
[image]

To select multiple objects in the 3D View, press the control key and click.

VexMex, 2021-12-23:
> I have stated the same thing before. It takes a bit of work, but you can use the merge function to avoid this problem. Select multiple objects, and then select "merge" from the right-click menu. [...]

that is very helpful! thanks!
And i must say ive never noticed that, last time i used rafts was long ago, but here its really obviously visible
i thought it would do the same as with a border, but seems like thats not the case

Anyways, thanks for the workaround!

rtyr, 2021-12-23:
It seems resolved.

Otherwise a duplicate of #4517, #2501, #4788. As of now, there is no collision detection of individual objects, it is up to the user to prevent them. 

Closing.

closed 2021-12-23 by rtyr