prusa3d/PrusaSlicer #7574: PrusaSlicer 2.4 does not open

Gabri988, 2021-12-23:
### PrusaSlicer 2.4
### Windows 10 on Bootcamp
The program starts and then closes.
I believe it is an incompatibility problem with the graphics card or OpenGL.
I tried to uninstall my video card drivers (ATI Mobility Radeon HD 4670) and the program works perfectly but is very slow due to the lack of graphics drivers, so it is unusable.
[image]

Koekoek3D, 2021-12-23:
Hi Gabri,

Just wanted to let you know that you are not the only person who is having troubles opening the latest version of PrusaSlicer 2.4.0... 

I have the exect same issue and I am also using Windows Home 10 21H2 (build: 19044.1415), but with an Intel(R) Core(TM) i7 CPU 920 @ 2.67GHz and Ati Radeon HD4800 series Graphics card. (also installed the latest version of the GC drivers, but these have not been renewed since 2013 :x)

Also note that I have been in contact with the helpdesk yesterday and all morning, unfortunetly without any luck.

Could you check if you can run beta1 of the 2.4.0, this was the last version which opend normally..?

bubnikv, 2021-12-23:
@Koekoek3D <https://github.com/Koekoek3D>

Are you also running on bootcamp? I suppose bootcamp is Windows on MAC
laptop / bare metal, not a virtual machine?

Koekoek3D, 2021-12-23:
No, not running Bootcamp.. It is just a regular computer.

bubnikv, 2021-12-23:
Would maybe help if you deleted or better renamed (backed up) your
configuration folder?

"C:\Users\your_user_name\AppData\Roaming\PrusaSlicer" or "C:\Documents and
Settings\your_user_name\Application Data\PrusaSlicer"

How exactly does it not start?

@Gabri988 <https://github.com/Gabri988>

*It is weird, that there is no text at the left column of the splash
screen.*

Koekoek3D, 2021-12-23:
Yes, deleted the configuration folder each time before I installed (or run a beta version) new version. (located at AppData\Roaming... cannot find the folder Application Data\..)

After clicking the program, the pop-up is exactly the same as Gabri988 posted.. it has no information on the left splash screen and once the splash screen and once it dissapears, nothing happends.

Gabri988, 2021-12-23:
> Hi Gabri988,
> 
> Just wanted to let you know that you are not the only person who is having troubles opening the latest version of PrusaSlicer 2.4.0...
> 
> I have the exect same issue and I am also using Windows Home 10 21H2 (build: 19044.1415), but with an Intel(R) Core(TM) i7 CPU 920 @ 2.67GHz and Ati Radeon HD4800 series Graphics card. (also installed the latest version of the GC drivers, but these have not been renewed since 2013 :x)
> 
> Also note that I have been in contact with the helpdesk yesterday and all morning, unfortunetly without any luck.
> 
> Could you check if you can run beta1 of the 2.4.0, this was the last version which opend normally..?

Hi, yes I have seen that there are other people who have the same problem as me.
Also I can no longer update the video card because it has been out of production for a long time.
Some time ago I tried the beta but I don't remember if it was the beta1 and that worked great

Gabri988, 2021-12-23:
> Would maybe help if you deleted or better renamed (backed up) your configuration folder? "C:\Users\your_user_name\AppData\Roaming\PrusaSlicer" or "C:\Documents and [...]

I have deleted that folder many times and it still doesn't work

Gabri988, 2021-12-23:
@bubnikv @Koekoek3D I confirm that the alpha1 version of prusaslicer 2.4 works correctly
[image]

Koekoek3D, 2021-12-23:
Kindly also verify that beta1 works and that the issue starts with beta2 (and onwards..)

Maybe we both are facing this issue because of the "old" ATI graphic card..

For whatever it is worth:
[image]

bubnikv, 2021-12-23:
@Koekoek3D

Is the beta1 the last version working for you?

These are dreaded issues. It may very well be an OpenGL driver issue,
however it is very hard to debug without having access to your computer.

We may try to bisect our builds to try to pinpoint the change that broke
PrusaSlicer for you, but it will be time consuming.

n8bot, 2021-12-23:
Could it be this commit? 4d2017188d5ddde8c218de1db97e8fec71876ad5

Gabri988, 2021-12-23:
@Koekoek3D Amd I have always hated it as a brand

KurtTrappschuh, 2021-12-24:
Hi,

I have exactly the same problem, no text on the splash screen und the programm closes after a couple of seconds.
Old PC, Core2Quad CPU, ATI Radeon HD 4300/4500.
Works fine with Beta 1, does not work with Beta 4 and newer (Beta 2 and 3 never tried).

But:

All Versions (including final 2.4) work fine if I start them with Administrator Priviledges !

Any ideas ...

Gabri988, 2021-12-24:
It doesn't work for me even if I start it with administrator privileges

I am hoping for a change from the developers, because if a beta version works I don't see why the final 2.4 shouldn't be done

sarvenn, 2021-12-25:
Beta 1 is the latest working version. Newer versions and final release does not open.
My previous PS version was 2.3.3 final release.

PrusaSlicer
Version:   2.4.0-beta1+win64
Build:     PrusaSlicer-2.4.0-beta1+win64-202110251333

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 4,293MB
OpenGL installation
GL version:   3.3.11672 Compatibility Profile Context
Vendor:       ATI Technologies Inc.
Renderer:     ATI Mobility Radeon HD 3650
GLSL version: 3.30
<details>
<summary>Installed extensions:</summary>
GL_AMDX_debug_output
GL_AMDX_vertex_shader_tessellator
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_name_gen_delete
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_sample_positions
GL_AMD_shader_stencil_export
GL_AMD_vertex_shader_tessellator
GL_ARB_ES2_compatibility
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_stencil_export
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_snorm
GL_ARB_texture_storage
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_draw_buffers
GL_ATI_envmap_bumpmap
GL_ATI_fragment_shader
GL_ATI_meminfo
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_buffer
GL_EXT_copy_texture
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_histogram
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texgen_reflection
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_buffer_object
GL_EXT_texture_compression_latc
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_IBM_texture_mirrored_repeat
GL_KTX_buffer_region
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_explicit_multisample
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

nafisca, 2021-12-26:
I have same problem on an older Dell XPS-420 running Windows 10 Pro. It has an ATI Radeon HD 2400 Pro and 2 quad cpu's q6600@2.4Ghz.

KurtTrappschuh, 2021-12-27:
I possibly forgot one important information:

Starting the installed Version 2.4.0 with Administrator Privileges does not work on my PC.
Only the "portable" Versions (from the zip files) start fine with "Run as Administrator". At least Beta4, RC2 and final 2.4.0.

Here my System Information:

PrusaSlicer
Version:   2.4.0+win64
Build:     PrusaSlicer-2.4.0+win64-202112211614

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 8,589MB
OpenGL installation
GL version:   3.3.11672 Compatibility Profile Context
Vendor:       ATI Technologies Inc.
Renderer:     ATI Radeon HD 4300/4500 Series
GLSL version: 3.30
<details>
<summary>Installed extensions:</summary>
GL_AMDX_debug_output
GL_AMDX_vertex_shader_tessellator
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_name_gen_delete
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_sample_positions
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_texture_cube_map_array
GL_AMD_texture_texture4
GL_AMD_vertex_shader_tessellator
GL_ARB_ES2_compatibility
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_stencil_export
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_snorm
GL_ARB_texture_storage
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_draw_buffers
GL_ATI_envmap_bumpmap
GL_ATI_fragment_shader
GL_ATI_meminfo
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_buffer
GL_EXT_copy_texture
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_histogram
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texgen_reflection
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_buffer_object
GL_EXT_texture_compression_latc
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_IBM_texture_mirrored_repeat
GL_KTX_buffer_region
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_explicit_multisample
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

bubnikv, 2021-12-27:
> Installed extensions: GL_AMDX_debug_output

Does PrusaSlicer start if you start it with the --datadir parameter
specifying an alternate location of the configuration folder? Namely, I
have a suspicion that for some of you accessing the configuration possibly
stored by the PrusaSlicer 2.4.0-alpha or beta into PrusaSlicer-alpha resp.
PrusaSlicer-beta folders fail in some unexpected way. If started with
--datadir parameter, copying from the -alpha or -beta folders is not
performed.

Please open a command line terminal and start prusa-slicer.exe with
--datadir parameter specifying some temp directory. Will it start or will
it crash the same way?

KurtTrappschuh, 2021-12-27:
That does not make a difference.
In a normal command line window it crashes the usual way.
In a command line window with Administrator Priveledges ist starts and then the configuration assistant shows up (like expected).

Is there a log file somewhere ?

bubnikv, 2021-12-27:
If you start prusa-slicer-console.exe with --loglevel=9 parameter, all
possible logs are spit out into the console.
You may try it, however I am not very optimistic it will help. We should
add a detailed logging of the start-up progress.

The best option as of now is to provide you guys with our development
builds between beta1 and beta2. Bisecting the development builds will help
us top pinpoint the offending change.

KurtTrappschuh, 2022-01-03:
Here the logfile.
First attempt in a normal console window, second in a console window with administrator priviledges.
Version 2.4.0 "portable".
First attempt ends after a couple of seconds,
Second attempt runs fine.
I hope, there is somethin in that you can use.

[Logfile.txt](https://github.com/prusa3d/PrusaSlicer/files/7801439/Logfile.txt)

bubnikv, 2022-01-03:
@KurtTrappschuh 

```
[2022-01-03 11:22:55.962341] [0x00001d8c] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): Das System kann die angegebene Datei nicht finden
```

This looks suspicious.

Do you think you can manage to monitor file openings with 
https://docs.microsoft.com/en-us/sysinternals/downloads/procmon
to find out on which the boost::filesystem::permissions fail?

Gabri988, 2022-01-03:
Today I was able to open prusa-slicer-console with admin privileges and it works perfectly

Gabri988, 2022-01-03:
I didn't understand how to generate a log

bubnikv, 2022-01-03:
These are some builds between beta1 and beta2. Would you please test which work and which don't? There were 250 commits between beta1 and beta2 and bisecting the interval helps us to identify the offending change.

https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta1+80-win64-g69b7fac01-202111080013.zip
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta1+131-win64-g8369bd86f-202111120013.zip
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta1+164-win64-g3c4095cec-202111151201.zip
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta1+210-win64-g1b7975639-202111220013.zip

PrusaSlicer-2.4.0-beta1+210-win64-g1b7975639-202111220013.zip

Koekoek3D, 2022-01-03:
Non of the 4 beta1 versions work for me.. only the [beta1 version](https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.4.0-beta1/PrusaSlicer-2.4.0-beta1+win64-202110251333.zip) which can be downloaded here.

bubnikv, 2022-01-03:
These are builds between beta1 and beta1+80. Would you please test these?

https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta1+15-win64-g145f0489b-202110270657.zip
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta1+25-win64-g93e91bcac-202110272316.zip
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta1+45-win64-g73af0ed33-202111020854.zip
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta1+63-win64-gc8ff0de27-202111040013.zip

Koekoek3D, 2022-01-03:
I can confirm that all of these beta1 (15/63) work!

KurtTrappschuh, 2022-01-03:
Hi bubnikv,

beta1+63 works the normal way,
beta1+80 only starts with administrator Priviledges

Somewhere between 63 and 80 ...

For the test with Process Monitor I will need a bit more time.

Thanks

bubnikv, 2022-01-03:
Thanks, thus the issue is between 64 and 80.

https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta1+65-win64-gaff9e1f7e-202111050938.zip
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta1+73-win64-gddef93e78-202111051510.zip
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta1+79-win64-g80695603d-202111052158.zip

KurtTrappschuh, 2022-01-03:
Hi bubnikv,

beta1+65 is not ok.

So, 64 must be the last working version or the first not working version.  ;-)

bubnikv, 2022-01-03:
Thanks. Thus it seems this commit broke PrusaSlicer for you guys:
aff9e1f7ea5cbaec4e9b259ad375ba73fc667a59

I am also compiling the 63 and 65 versions once again, as the issue may be not in PrusaSlicer source code base, but in our wxWidgets code.

bubnikv, 2022-01-03:
Our wxWidgets patch associated with the change between 63 and 65 is 4b7588eb0fec2d3e9b48e8b1d27e347ba91011a8
Both commits are relatively bulky. It will not be quite easy to debug the issue.

Gabri988, 2022-01-03:
@bubnikv 
PrusaSlicer-2.4.0-beta1+80-win64-g69b7fac01-202111080013 not working
PrusaSlicer-2.4.0-beta1+210-win64-g1b7975639-202111220013 working with amministrator privileges
PrusaSlicer-2.4.0-beta1+164-win64-g3c4095cec-202111151201 working with amministrator privileges
PrusaSlicer-2.4.0-beta1+131-win64-g8369bd86f-202111120013 working with amministrator privileges
PrusaSlicer-2.4.0-beta1+63-win64-gc8ff0de27-202111040013 normally working 
PrusaSlicer-2.4.0-beta1+25-win64-g93e91bcac-202110272316 normally working
PrusaSlicer-2.4.0-beta1+15-win64-g145f0489b-202110270657 normally working
PrusaSlicer-2.4.0-beta1+45-win64-g73af0ed33-202111020854 normally working
PrusaSlicer-2.4.0-beta1+79-win64-g80695603d-202111052158 working with amministrator privileges

bubnikv, 2022-01-03:
Do you guys think you can track the Win32 API calls and their failure codes with

https://docs.microsoft.com/en-us/sysinternals/downloads/procmon

and provide us with the logs? We hope to find the Win32 API call that fails if not run in administrator mode.

I think it makes the most sense to test with the final. Thank you.

bubnikv, 2022-01-03:
Capturing the Win32 API calls with procmon is likely the easiest way to debug the issue. If that does not work, we will have to produce a special build with logging the application start up progress to find out what offenses your Windows. 

By the way I just created a non-admin account on my Windows 11 box and PrusaSlicer starts just fine there.

KurtTrappschuh, 2022-01-03:
Hi bubnikv,
[Logfile.zip](https://github.com/prusa3d/PrusaSlicer/files/7803290/Logfile.zip)

that is done with 2.4.0 with    prusa-slicer-console.exe --loglevel=9    (to have timestamps).
A couple of secons before I started recording with ProcessMonitor.

I hope that helps a bit ...

Thanks

Gabri988, 2022-01-03:
> 
can I ask you the procedure on how you generated the logfile on winodws? thank you

Gabri988, 2022-01-03:
@KurtTrappschuh can I ask you the procedure on how you generated the logfile on winodws? thank you

Excuse my ignorance

KurtTrappschuh, 2022-01-03:
Hi Gabri988,

I am not sure if I recorded the data that bubnikv wants.
But let's try:

- download ProcessMonitor
- unzip it
- click with right mouse button on "Procmon64.exe" and "Start as Administrator"
- you immideately get thousands of lines per second
- deactivate recording with the "Capture" Icon (3rd from left)
- delete the list with the "Clear" Icon

- prepare the Start of Prusa Slicer 2.4.0
- start "Capture" and start Prusa Slicer
- when Prusa Slicer is gone stop "Capture"
- you now can clean the list from unwanted entries with right click "exclude 'xxx.xxx'" (optional)
- "Save" (Native Process Monitor Format), I hope that is ok (I did not change the default settings)

Again, I am not sure if that produces what bubnikv wants.

Have fun ;-)

n8bot, 2022-01-03:
@bubnikv perhaps this is related to this: https://github.com/prusa3d/PrusaSlicer/issues/7546

Since it is related to dark mode. It seems to be the NppDarkMode::InitDarkMode() call that is causing the problem.

That is exactly the function, iirc, that got broken with the 21H2 update. Compile a version with the patched wxwidgets found here: https://github.com/n8bot/PrusaSlicer/releases/download/wxpatchtest/v3.1.4-n8patched.zip

I definitely could be wrong but it seems all related with all the info.

n8bot, 2022-01-04:
Upon closer inspection. it seems very unlikely that the simple version check fix will alleviate this problem. Either the if statement returns true or not -- there should be no reason the app would crash merely for that.

bubnikv, 2022-01-04:
@KurtTrappschuh 

Thanks Kurt for the log.

At the end of the day, we were able to reproduce the issue on one Windows 11 computer when connected over remote desktop. It does exactly what you have described, it crashes when opening a splashscreen. If executed with administrator rights, it does not crash.

It crashes due to recursive call somewhere in the new Windows dark mode support. We are on it.

Gabri988, 2022-01-04:
Thanks you very much @KurtTrappschuh for helping me generate the log file.
[Logfile prusaslicer.zip](https://github.com/prusa3d/PrusaSlicer/files/7806930/Logfile.prusaslicer.zip)

@bubnikv 
I updated my windows 10 home to the PRO version. And I found that when working remotely from the pc, prusaslicer 2.4 final starts up without problems

bubnikv, 2022-01-04:
@YuSanka will provide you soon with a PrusaSlicer.dll to test whether it fixes your start-up issue. We believe it should work.

YuSanka, 2022-01-04:
Please try to replace this DLL in your "...\PrusaSlicer-2.4.0+win64" folder and run the "prusa-slicer.exe"
[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/7806986/PrusaSlicer.zip)

Gabri988, 2022-01-04:
Now works! Thank you so much guys @YuSanka @bubnikv

KurtTrappschuh, 2022-01-04:
It also works fine on my PC, on the "portable" Version and also on the installed Version.

Many thanks for all who were involved in this fix.
Kurt

Koekoek3D, 2022-01-04:
Yeah!! Great work guys, it works perfectly =)

Thanks a million!

n8bot, 2022-01-04:
Holy crap I was right! Well, I mean, partially... on my second guess. lol.

fpikus, 2022-01-04:
I redownloaded the installer and it ran fine. So did the installer I downloaded from prusa3d.com

nafisca, 2022-01-05:
Thanks! That worked for me!

From: Oleksandra Yushchenko
Sent: Tuesday, January 4, 2022 5:43 AM
To: prusa3d/PrusaSlicer
Cc: nafisca; Comment
Subject: Re: [prusa3d/PrusaSlicer] PrusaSlicer 2.4 does not open (Issue #7574)

Please try to replace this DLL in your "...\PrusaSlicer-2.4.0+win64" folder and run the "prusa-slicer.exe"
PrusaSlicer.zip

tobiaslive, 2022-01-06:
> Please try to replace this DLL in your "...\PrusaSlicer-2.4.0+win64" folder and run the "prusa-slicer.exe" [PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/7806986/PrusaSlicer.zip)

yes it's works

sarvenn, 2022-01-08:
> Please try to replace this DLL in your "...\PrusaSlicer-2.4.0+win64" folder and run the "prusa-slicer.exe" [PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/7806986/PrusaSlicer.zip)

Worked on me as well. Thanks mate.

bubnikv, 2022-01-27:
Fixed with https://github.com/prusa3d/wxWidgets/commit/73f029adfcc82fb3aa4b01220a013f716e57d110
The fix will be released with PrusaSlicer 2.4.1-beta1.

closed 2022-01-27 by bubnikv

Rauchbier87, 2022-12-29:
V2.5.0 doesn't seem work on windows either

stepikovo, 2023-01-03:
@Rauchbier87 please provide us with more information, screenshots, error messages, system info etc.