prusa3d/PrusaSlicer #7575: [Feature Request]

federico9292, 2021-12-23:
Hi! I'd like to request a feature, I don't know if and how hard it can be to implement. 

You know that there are filaments in the market with co-extruded colors, meaning that half section of the filament is of one color, the other half is another
[image]
meaning that the direction the nozzle moves, affects the color that will be shown. 
[image]
[image]
[image]
 
This implies that if you can selectively choose which direction the layer should be printed you can tune the results showing just one color on one side (this vase is shown through a mirror, what a cool effect) 
[image]

I'm here to request in there is a possibility to chose something like the direction on the printing, 

like a check if the nozzle is moving from X10 to X20 as an increasing value and there is an option checked that only movements from left to right are validated, then that movement from X10 to X20 it's made in without extruding and the same movement is replicated but from right to left while extruding.

foreachthing, 2021-12-23:
How do you ensure the correct position of the filament (and to prevent it from rotating? ) while printing?

federico9292, 2021-12-23:
> 

I actually don't know how this dark magic works, in fact I do not even own a coextruded filament to try. I assume that the grip from the extruder motor is enough to prevent it from rotating on it's axis..

degroof, 2022-10-04:
I think the colors would work better if the fill pattern was changed to concentric.

[image]

[image]

federico9292, 2022-10-04:
Yeah that I figured (I bought 4 coextruded filaments) altough the most of the visible effect is through the "External Perimeter" characteristic

degroof, 2022-10-04:
Yep. Perimeters work fine regardless, but there's a definite difference in top surfaces between monotonic and concentric. I just ran a couple test prints using red-yellow-blue coextruded. The outer walls are roughly the same but the upper surface of the concentric one is much more visually appealing.

[image]

degroof, 2022-10-04:
> How do you ensure the correct position of the filament (and to prevent it from rotating? ) while printing?

With a direct drive extruder, at least, the filament seems to stay aligned with how it's spooled, with minimal twisting. Not sure if a Bowden extruder would be as consistent, though.

federico9292, 2022-10-04:
as a bowden user I can confirm that the filament with the bowden tube does not twist a lot. my initial question was regarding forcing one direction for movements for perimeters, like I want only the yellow to show on that side -> forced movement of the printer in "that" direction. but I suppose it's a bit tricky and not easily implementable

degroof, 2022-10-04:
Ah, gotcha. So, you'd need some sort of unidirectional fill pattern, where the extruder lays down filament in a single direction, then travels back (without extruding) to start the next run. I suppose the angle would need to be specified as well, so you could select the color mix.

PTrUNB, 2025-09-24:
[image]

Here's another example that would this feature a good thing

Rather frustrating to need another slicer on the Prusa line up.

closed 2026-09-29 by prusaslicer-bot[bot]