prusa3d/PrusaSlicer #7576: Arrange doesn't work properly (parts are arranged in each other)

Tlevendig, 2021-12-23:
### Version
_2.4.0 (beta 3 and final release)_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Windows 11_

### 3D printer brand / version + firmware version (if known)
_Default Voron 0.1 profile, but it also happens with the other default profiles (like Prusa Mini and MK3S) _

### Behavior
* Two STL files are arranged in each other after using the arrange feature, as seen on this screenshot:
* 
[image]

* _Steps needed to reproduce the problem_
* Load the attached STL file, and add one more instance, then use arrange feature. 

* _Expected Results: parts arranged next to each other_
*
* _Actual Results: parts arranged in overlapping manner_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
  * 
[image]
[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[Prusaslicer arrange problem.zip](https://github.com/prusa3d/PrusaSlicer/files/7769034/Prusaslicer.arrange.problem.zip)
[STL.zip](https://github.com/prusa3d/PrusaSlicer/files/7769035/STL.zip)

bubnikv, 2021-12-23:
Cannot reproduce, it seems to work just fine to me.

[image: image.png]

lukasmatena, 2023-10-06:
Closing as outdated. It works fine in 2.6.1.

closed 2023-10-06 by lukasmatena