prusa3d/PrusaSlicer #7577: Creality print settings gone after update 2.4.0

dsagdfgsFD, 2021-12-23:
### Version
_Version of PrusaSlicer used goes here_
2.4.0
_Use `About->About PrusaSlicer` for release versions_
Version 2.4.0+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Creality CR-6 SE
Prusa MK3S+

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

After update, the print settings for my Creality CR 6 SE is gone. This is only for the user preset with a 0.6mm nozzle. I also have a user preset with 0.4 mm nozzle where i am able to choose print setting premade for creality. 

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-12-23:
What was your previous PrusaSlicer version? Was it some alpha or beta release? Alpha/beta releases are using different data directory than final.

rtyr, 2022-06-20:
No response. Closing.

closed 2022-06-20 by rtyr