prusa3d/PrusaSlicer #7580: If Top Layers is set to 0, there is no infill; Will not create less than 4 Top Layers

Jam-ehz, 2021-12-23:
### Version
Build: PrusaSlicer-2.4.0+win64-202112211614

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 (build 19044), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa Mini+

### Behavior
Prior to 2.4, adding a modifier object with 0 Top and Bottom Layer would expose just the infill pattern to be printed.
Now, 2.4 with Top Layers = 0 in the modifier produces no infill. Found this same behaviour when setting Top Layers to 0 in the Print Settings as well.

Also, Setting Top Layers in the modifier or Print Settings to less than 4 layers will still produce 4 layers.
Just checked now, same for Bottom Layers - Won't make less than 3 Bottom Layers either.

#### Project File (.3MF) where problem
[image]
[image]
[image]
[Prusa Infill 3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7770347/Prusa.Infill.3mf.zip)

liftbag, 2021-12-23:
I think the missing infill with 0 top layer is just a bug. 
As for the second behavior, it is perfectly normal. You have set the top minimum shell thickness to 0.7mm, therefore, if you enable the top layers (even only one), 0.7mm thickness must always be guaranteed. With 0.2mm layer height, 3 top layers would not be enough, so the minimum possible is 4. To restore total control over the number of top and bottom layers, you need to set top and bottom minimum shell thickness to 0.

Jam-ehz, 2021-12-23:
Ah, gotcha. It's not necessarily printed layers,  but physical layers.
And I think I figured out the first problem, the setting "Only Infill When Needed" was checked.

closed 2021-12-23 by Jam-ehz