prusa3d/PrusaSlicer #7581: Perimeter Modifier produces no perimeter when set to 2.

Jam-ehz, 2021-12-23:
Build: PrusaSlicer-2.4.0+win64-202112211614

Use About->About PrusaSlicer for release versions

For -dev versions, use git describe --tag or get the hash value for the version you downloaded or git rev-parse HEAD

Operating system type + version
Windows 10 (build 19044), 64-bit edition

3D printer brand / version + firmware version (if known)
Prusa Mini+

### Behavior
When modifier perimeter is set to the same value as in Print Settings, no perimeters are made in the affected area.

Screenshots show mod perimeter set at 1, 2, and 3 when Print Settings Perimeter = 2. Same thing happens when you set both mod and print settings perimeter to 1 or 3 ir niore.

Also note the lack of infill, that's another issue I had just posted, That happens when Top Layer = 0 in the modifier OR the Print Settings.

### .3mf and Attached Files
[Prusa Perimeter 3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7770421/Prusa.Perimeter.3mf.zip)
[image]
[image]
[image]
[image]

bubnikv, 2021-12-23:
Behavior of modifiers changed in
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1 ,
please read the section on modifiers and their order.

Many of the changes were done to support multiple extruders / colors
correctly. Namely, we now support one modifier modifying parameters of
another modifier, for example one modifier sets the infill and the other
modifier overrides the extruder color.

Now the framework tries to be smart and if it finds out that the modifier
does not override any parameter of any of the modifiers before in object
list, it is not applied. Thus if you apply a modifier with the same
parameters including the number of perimeters, the region will not be split.

Is it a problem?

n8bot, 2021-12-23:
@bubnikv I agree that the new modifier behaviour seems more useful than before (however, I have never and don't currently use them, so my opinion means less than someone who uses the feature often).

However, I've found another problem with the new algorithm. I noticed the other day that the modifier was making a new region even though NONE of the settings in the modifier were changed.

It appears that, with a system PRINTER profile, combined with some USER print profile, the modifier will always make a new region? That may not be the exact (or only) combination that causes this, but it seems to be a reliable way to trigger it.

[image]

n8bot, 2021-12-23:
Actually, you don't even need to use a SYSTEM printer profile. If you just use any user profile, the modifiers always create a new region it seems?

It seems that the comparison to profiles to see if anything has changed is very SYSTEM profile-centric at the moment.

bubnikv, 2021-12-23:
@n8bot I cannot reproduce your issue.

[image: image.png]

bubnikv, 2021-12-23:
[image]

bubnikv, 2021-12-23:
@n8bot could you please provide a 3MF for investigation? Thanks.

n8bot, 2021-12-23:
Yes just a second

n8bot, 2021-12-23:
[Error.zip](https://github.com/prusa3d/PrusaSlicer/files/7771741/Error.zip)

@bubnikv

n8bot, 2021-12-23:
I see a potential element at play: the profile you used were based on system profiles, while mine are from who-knows-where.

So maybe the fact that a profile is previously from a system profile affects something. Also, maybe a weirdly-made user profile causes the problem.

bubnikv, 2021-12-23:
@n8bot Cannot reproduce with the 3MF "error" you provided.

n8bot, 2021-12-23:
That is weird. I'm loading it into a bone-stock 2.4.0 compiled by you (downloaded from GH), empty configuration folder, with no other profiles present except the default.

n8bot, 2021-12-23:
Also, make sure after you slice you preview down the layers a bit -- the modifier is only present in the middle of the moddle, not near the top or bottom.

n8bot, 2021-12-23:
Step 1: Open bare, new, no-existing configuration PrusaSlicer 2.4.0:
[image]

Step 2: Load error.3mf (click no when asked to change support/bridge settings):
[image]

Step 3: Slice and scroll the preview down to near the middle of the layers range:
[image]

bubnikv, 2021-12-23:
@n8bot Thanks I can see it now. This needs to be investigated under a
debugger, however I will do it first after the new year.

n8bot, 2021-12-23:
I'm not very familiar with the sections of code that deal with this, but I might find some time to play around and if I discover anything I'll post here (without tagging you, so hopefully you can enjoy your family without thinking of PrusaSlicer over the holidays!)

Have a nice one!

closed 2026-09-29 by prusaslicer-bot[bot]