prusa3d/PrusaSlicer #7583: PrusaSlicer 2.4 crashes when I quit

epetit91, 2021-12-23:
### Version
PrusaSlicer 2.4

### Operating system type + version
MacOS 10.12.6 Sierra

### 3D printer brand / version + firmware version (if known)
PrusaSlicer I3 MK3S

### Behavior
Prusaslicer 2.4 always crashes when I quit the application.

[image]

bubnikv, 2021-12-23:
Would you please post the detailed crash report? It shall open with one of
the buttons of the dialog you have posted, I don't know which one of the
buttons, I don't understand French.

epetit91, 2021-12-23:
Detailed crash report :

Process:               PrusaSlicer [1444]
Path:                  /Volumes/VOLUME/*/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.4.0+x64-202112211607 (PrusaSlicer-2.4.0+x64-202112211607)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           PrusaSlicer [1444]
User ID:               501

Date/Time:             2021-12-23 19:41:30.906 +0100
OS Version:            Mac OS X 10.12.6 (16G2136)
Report Version:        12
Anonymous UUID:        F3B6566F-276B-9E81-EAD2-2083251BFABE

Time Awake Since Boot: 3700 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000009df000003a8
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0x9df000003a8:
    CG backing stores      000000011a0c3000-000000011ae90000 [ 13.8M] rw-/rw- SM=PRV  
--> 
    MALLOC_NANO            0000600000000000-0000600002800000 [ 40.0M] rw-/rwx SM=PRV  

Application Specific Information:
objc_msgSend() selector name: closing_update:

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	0x00007fff9d43c01d objc_msgSend + 29
1   com.apple.CoreFoundation      	0x00007fff8814cfbc __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 12
2   com.apple.CoreFoundation      	0x00007fff8814cf4f ___CFXRegistrationPost_block_invoke + 63
3   com.apple.CoreFoundation      	0x00007fff88156c9c __CFRUNLOOP_IS_CALLING_OUT_TO_A_BLOCK__ + 12
4   com.apple.CoreFoundation      	0x00007fff88137e34 __CFRunLoopDoBlocks + 356
5   com.apple.CoreFoundation      	0x00007fff881375a5 __CFRunLoopRun + 917
6   com.apple.CoreFoundation      	0x00007fff88136fb4 CFRunLoopRunSpecific + 420
7   com.apple.HIToolbox           	0x00007fff87695ebc RunCurrentEventLoopInMode + 240
8   com.apple.HIToolbox           	0x00007fff87695bf9 ReceiveNextEventCommon + 184
9   com.apple.HIToolbox           	0x00007fff87695b26 _BlockUntilNextEventMatchingListInModeWithFilter + 71
10  com.apple.AppKit              	0x00007fff85c2aa04 _DPSNextEvent + 1120
11  com.apple.AppKit              	0x00007fff863a67ee -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 2796
12  com.apple.AppKit              	0x00007fff85c1f38b -[NSApplication run] + 926
13  com.prusa3d.slic3r/           	0x0000000102f5830e wxGUIEventLoop::OSXDoRun() + 174
14  com.prusa3d.slic3r/           	0x000000010304172d wxCFEventLoop::DoRun() + 29
15  com.prusa3d.slic3r/           	0x0000000102fb8955 wxEventLoopBase::Run() + 85
16  com.prusa3d.slic3r/           	0x0000000102f90bd4 wxAppConsoleBase::MainLoop() + 116
17  com.prusa3d.slic3r/           	0x0000000102f1b25a wxApp::OnRun() + 26
18  com.prusa3d.slic3r/           	0x0000000102fe9203 wxEntry(int&, wchar_t**) + 131
19  com.prusa3d.slic3r/           	0x0000000102282c5c Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 236
20  com.prusa3d.slic3r/           	0x0000000101a2131f Slic3r::CLI::run(int, char**) + 18767
21  com.prusa3d.slic3r/           	0x0000000101a25e90 main + 352
22  libdyld.dylib                 	0x00007fff9dd7f235 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff9deae44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff9df98621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff9df9807d start_wqthread + 13

[... 328 lines cut ...]
Serial ATA Device: WDC WD1001FALS-40Y6A0, 1 TB
Serial ATA Device: HL-DT-STDVDRW  GA32N
Serial ATA Device: CT1000MX500SSD1, 1 TB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Keyboard Hub
USB Device: Apple Keyboard
USB Device: USB2.0 Hub
USB Device: USB 2.0 Hub
USB Device: USB3.0 External HDD
USB Device: Ext. HDD
USB Device: Card Reader
USB Device: BRCM2046 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Xslim3.0
USB Device: IR Receiver
USB Device: Built-in iSight
FireWire Device: OXFORD IDE Device 1, Oxford Semiconductor Ltd., Up to 400 Mb/sec
FireWire Device: OXFORD IDE Device 1, Macpower, Up to 400 Mb/sec
FireWire Device: OXFORD IDE Device 1, Macpower, Up to 400 Mb/sec
Thunderbolt Bus: 

Eric

bubnikv, 2021-12-23:
Thanks. I am not sure whether it is specific to your OSX Sierra, which is
quite old. We may try to replicate your issue on a virtual machine with
Sierra after Christmas.
Interestingly nobody else reported this issue.

epetit91, 2021-12-23:
All was OK with previous version 2.3.3.

epetit91, 2021-12-24:
Hello,
today I tried PrusaSlicer 2.4 with a Macbook Pro running 10.12.6 Sierra and I see the same problem.
I also tried in High Sierra 10.13.6 and everything works fine.
So this is indeed a Sierra problem.

epetit91, 2021-12-25:
Hello and Merry Christmas everyone.
With VMWARE Fusion I installed an OSX Sierra virtual machine and tried PrusaSlicer 2.4 and the result is the same: when I quit PrusaSlicer the application crashes.

bubnikv, 2021-12-25:
Thanks, we will try to look into it after Holidays.

andy9a9, 2021-12-25:
I have the similar problem with adding/removing filament. I've just opened/close dialog to add filament and it crashed.
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [8791]
Path:                  /Applications/Original Prusa Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.4.0+x64-202112211607 (PrusaSlicer-2.4.0+x64-202112211607)
Code Type:             ARM-64 (Native)
Parent Process:        zsh [8599]
Responsible:           Terminal [7604]
User ID:               501

Date/Time:             2021-12-25 13:16:06.9655 +0100
OS Version:            macOS 12.1 (21C52)
Report Version:        12
Anonymous UUID:        49AD0CDB-21D2-E759-2D53-59650567835C

Sleep/Wake UUID:       B72E38CA-0904-4E7B-939E-EB31D475A5DA

Time Awake Since Boot: 12000 seconds
Time Since Wake:       4629 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000037
Exception Codes:       0x0000000000000001, 0x0000000000000037
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [8791]

VM Region Info: 0x37 is not in any region.  Bytes before following region: 4306452425
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100af4000-102240000    [ 23.3M] r-x/r-x SM=COW  ...S/PrusaSlicer

Kernel Triage:
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   PrusaSlicer                   	       0x1013ed360 bool std::__1::operator!=<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > >(std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > > const&, std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > > const&) + 48
1   PrusaSlicer                   	       0x1013ec7b4 Slic3r::GUI::ConfigWizard::priv::apply_config(Slic3r::AppConfig*, Slic3r::PresetBundle*, Slic3r::PresetUpdater const*, bool&) + 6008
2   PrusaSlicer                   	       0x1013ec7b4 Slic3r::GUI::ConfigWizard::priv::apply_config(Slic3r::AppConfig*, Slic3r::PresetBundle*, Slic3r::PresetUpdater const*, bool&) + 6008
3   PrusaSlicer                   	       0x1013ef538 Slic3r::GUI::ConfigWizard::run(Slic3r::GUI::ConfigWizard::RunReason, Slic3r::GUI::ConfigWizard::StartPage) + 448
4   PrusaSlicer                   	       0x10121c544 Slic3r::GUI::GUI_App::run_wizard(Slic3r::GUI::ConfigWizard::RunReason, Slic3r::GUI::ConfigWizard::StartPage) + 148
5   PrusaSlicer                   	       0x101e0e58c wxEvtHandler::ProcessEventLocally(wxEvent&) + 280
6   PrusaSlicer                   	       0x101e0e32c wxEvtHandler::ProcessEvent(wxEvent&) + 56
7   PrusaSlicer                   	       0x101e0e134 wxEvtHandler::ProcessPendingEvents() + 192
8   PrusaSlicer                   	       0x101d789e8 wxAppConsoleBase::ProcessPendingEvents() + 112
9   PrusaSlicer                   	       0x101e168b0 wxCFEventLoop::OSXCommonModeObserverCallBack(__CFRunLoopObserver*, int, void*) + 84
10  CoreFoundation                	       0x1c674be4c __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
11  CoreFoundation                	       0x1c674bc98 __CFRunLoopDoObservers + 572
12  CoreFoundation                	       0x1c674b1d0 __CFRunLoopRun + 764
13  CoreFoundation                	       0x1c674a734 CFRunLoopRunSpecific + 600
14  HIToolbox                     	       0x1cf2dcf68 RunCurrentEventLoopInMode + 292
15  HIToolbox                     	       0x1cf2dcbf4 ReceiveNextEventCommon + 320
[... cut ...]
18  AppKit                        	       0x1c929f584 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1332
[... 333 lines cut ...]
        "SIRI_DICTATION_ASSETS" : "61ae8d06da72d16a4beb762e"
      },
      "deploymentId" : 240000290
    },
    {
      "rolloutId" : "60da5e84ab0ca017dace9abf",
      "factorPackIds" : {

      },
      "deploymentId" : 240000008
    },
    {
      "rolloutId" : "607844aa04477260f58a8077",
      "factorPackIds" : {
        "SIRI_MORPHUN_ASSETS" : "6103050cbfe6dc472e1c982a"
      },
      "deploymentId" : 240000066
    }
  ],
  "experiments" : [

  ]
}
}
```

bubnikv, 2021-12-26:
> Andrej Valek:
> I have the similar problem with adding/removing filament. I've just
opened/close dialog to add filament and it crashed.

Your issue is most likely different from the original one. Are you running
the same OSX version as the original reporter? We may need your
configuration folder and a precise description of the steps you follow to
reproduce your issue.

andy9a9, 2021-12-26:
Sorry, I don't wanted to create a new one. Ok so,...
OSX: Monterey 12.1
PC: MacBook Pro 14.2 M1
Reproducer: Open the dialog to add new filament and click on Finish.

epetit91, 2021-12-26:
Hello,
for your information, I tried to add a new filament on MacOS 10.12.6 Sierra and I have the same problem as andy9a9 with the same message as my first post !!!

bubnikv, 2022-01-04:
@kocikdav please investigate whether this crash on closing the application on Sierra may be caused by 15d9751919c6937e537fb85f2dca63fca2aaa892

bubnikv, 2022-01-07:
The crash on closing is fixed with https://github.com/prusa3d/PrusaSlicer/commit/747cf8a0fe21545a6bf06171cb79c8c8bfb7ba7b
The fix will be released with PrusaSlicer 2.4.1.

For the crash in Wizard, please continue discussion at https://github.com/prusa3d/PrusaSlicer/issues/7655

closed 2022-01-07 by bubnikv