prusa3d/PrusaSlicer #7584: PrusaSlicer 2.4 skirt loops count

gjelsoe, 2021-12-23:
It seems like the Loops count for skirt is set for 2 loops as minimum, can somebody confirm this?

Tlevendig, 2021-12-23:
@gjelsoe For me it depends, when using the stock MK3S profile, changing the skirt count works fine (I can set it at whatever I want, and it also shows that amount in the GCODE preview. But when using EG the voron profile, it's at least 3 (or zero, that works as well)

rtyr, 2021-12-23:
No, it just also depends on minimal extrusion length.

[image]

Can we close the issue?

gjelsoe, 2021-12-23:
@rtyr Thanks, didn't notice that :)

Closing.

closed 2021-12-23 by gjelsoe