prusa3d/PrusaSlicer #7587: Place on Face operation takes a long time to complete

laptop-laptop, 2021-12-23:
### Version
2.4.0

_Use `About->About PrusaSlicer` for release versions_
2.4.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
i3MK3S+

### Behavior
Just like previous versions, loading of complex STL is very fast but when I clicked "Place on Face", depending on the STL sometimes it took a very long time to complete the task with '(No Responding)' showing up on top of the dimmed PS window and a spinning blue circle. For example, it took about 9 minutes for the 'Place on Face' operation to complete on a STL (100K triangles, 45M bytes) using an i9 PC with 64GB RAM. During the operation, task manager showed that PrusaSlicer was using about 10% of CPU and memory.

Tried the following method but there is no 'Set Priority' option as shown in Step.3
https://www.mobigyaan.com/allocate-more-ram-specific-app-windows-10

Also, tried 'Adjust for Best Performance' in System Properties' Settings but it does not improve.
https://www.techwalla.com/articles/how-to-give-programs-more-ram

If we can allocate more memory to PrusaSlicer, will the wait time be shortened? If so, please show us how to do it. 

Thank you

bubnikv, 2021-12-23:
> For example, it took about 9 minutes for the 'Place on Face' operation to complete on a STL

9 minutes is certainly a lot. We never experienced such thing. Would you please provide a model where the place on face takes so long?

evrenesat, 2021-12-24:
I'm experiencing a similar long wait for place on face with this file or with other parts of this design.

https://www.thingiverse.com/thing:4169512/files

[Porte_crayon_tux_V2-Pieds.STL.zip](https://github.com/prusa3d/PrusaSlicer/files/7771843/Porte_crayon_tux_V2-Pieds.STL.zip)

bubnikv, 2022-01-03:
Thanks, reproduced. We will take care of the unintended quadratic
complexity.

lukasmatena, 2022-01-03:
Hopefully fixed with b034402 on master branch. The problem was there since the first implementation of the Place on face tool 3.5 years ago. The fix will appear in the next release. Thanks.

bubnikv, 2022-01-27:
Fixed in the upcoming PrusaSlicer 2.4.1-beta1

bubnikv, 2022-01-27:
also improved with a554f1fb089ccbc6fee995571091054703ca181d

closed 2022-01-27 by bubnikv