prusa3d/PrusaSlicer #7588: Typo in new help feature in 2.4.0

rdragonrydr, 2021-12-23:
### Version
_Version of PrusaSlicer used goes here_
Version: 2.4.0+linux-x64
Build: PrusaSlicer-2.4.0+linux-x64-202112211614

### Operating system type + version
Ubuntu 18.04 64 Bit

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa i3 MK3S

There is a typo in the new help feature when I opened the Slicer after updating from the previous release.

[image]

n8bot, 2021-12-24:
You a word or two there!!! 😁

dartrax, 2022-01-01:
Another one, space before instead of after the dot.
[image]

closed 2022-01-17 by YuSanka

dartrax, 2022-03-05:
Another one: In the help text, "Von Festplatte neu laden" is used, the upper case "Von" is not correct. One might want to argue that it is a quotation of the context menu, but when looking at that, in the context menu it is written "Neuladen von Festplatte". So two mistakes actually.

Correct would be:

> Klicken Sie mit der rechten Maustaste auf das Modell in der 3D-Ansicht und wählen Sie Neuladen von Festplatte.

commit 2cc20594 (2022-01-17): Localization: Updated POT
+ Fixed some phrases in Preferences and Hints
+ fixed #7588 - Typo in new help feature in 2.4.0