prusa3d/PrusaSlicer #7589: Reload from disk function not working with ver 2.3.3 and 2.4.0

lcemoore, 2021-12-23:
### Version
2.4.0

### Operating system type + version
Windows 10 build, 21H2

### 3D printer brand / version + firmware version (if known)
MK3S+,
FW 3.1.0 4481

### Behavior
Since upgrading to PS 2.3.3 and now with 2.4.0, the reload from disk function does nothing.  There is no error message, and the old version of the object is not replaced.

#### Project File (.3MF) where problem occurs
This problem is not related to any specific file, and it does apply to both .3mf and .stl files

bubnikv, 2021-12-23:
We tested reload from disk extensively for PrusaSlicer 2.4.0 release after we fixed some issues of PrusaSlicer 2.3.3. Namely, we fixed reload from disk after a project is loaded from a 3MF. Reload from disk worked for a fresh project, but for a project loaded from 3MF the "reload from disk" feature did not reload a geometry at the same position and scale
#5541.

When the "Reload from disk" action was called on an object for which only the source file name, not the full path to the source file name was known, PrusaSlicer used to always ask the user to look up the source file location manually, while newly PrusaSlicer searches for the source file next to the active project file before asking the user for help.

In addition to reloading an existing object or volume geometry from disk, PrusaSlicer newly allows to replace the geometry with a different file
#3856.

This change log entry is a bug fix of an error introduced between 2.3.3 and 2.4.0 final: Fixed 'Reload from disk' which did not work for modifier volumes and which sometimes shifted the reloaded object incorrectly #7429.

However @jmerc77 complains at #7429 that reload from disk does not work for him, while three people of our team were not able to confirm @jmerc77 's findings. Frankly I don't know what to do. It works for us.

lcemoore, 2021-12-24:
It's not a show stopper for me, but it is odd. I can see why shifting the position upon reload would be annoying, but I never experienced that because reload from disk hasn't worked for me since the move to .3mf.

Thanks for your reply.

bombela, 2022-01-11:
For me, the reload overrides my parts with the first one in the file (first as in, the one physically first one the file). I have a .amf with two independent part. I can import this file, and I get the two parts in prusa slicer. Each part gets its unique name as well. The same name as was given within the amf file. When I refresh, part 1 and part 2 are replaced by the new part 1. They do keep theirs original names. But the geometry was replaced on both by the first part.

Took me a while to realize because I happen to have very similar parts (one slides into the other). And I couldn't figure why on earth my two parts had the exact same dimensions when printed! (I am printing with a new 0.8mm nozzle, I thought I had done something wrong).

Heckw, 2022-02-02:
For projects created with 2.4.0 reload from disk still doesn't work using 2.4.1 beta1.
For projects created with 2.4.1 beta1 reload from disk seems to work ok.

FrHePo, 2022-02-02:
Update: For all earlier created projects (before 2.4.1 beta1) reload from disk doesn't work!
Since I have some older projects with complicated modifiers this definitely doesn't make me happy.

bombela, 2022-02-02:
Also for me, reload from disk only work as long as I keep prusaslicer open. Here are the steps to reproduce:
 - open prusaslicer
 - import the different AMF each with a single uniquely named part
 - reload from disks works fine
 - save project to .3mf and close prusaslicer
 - open prusa slicer and recall project
 - reload from disk fails with a generic error (cannot reload from disk)

Storing the full path to the original AMF files within the .3mf doesn't change anything.

FrHePo, 2022-02-02:
Did you try the latest version (2.4.1 beta1)?

bombela, 2022-02-03:
On 2.4.1 beta1, replace with STL is grayed out for me. I was opening a file made with the stable version. Reload from disk fails with the same error. Replace with STL is disabled. So in some ways, the situation is worse for me.

bombela, 2022-02-03:
Reducing the number of replica to 1, enabled back the replace from STL. I replaced with a .AMF containing a single part. Then I saved. Closed prusa slicer 1.4.1 beta1. Open prusa slicer again and my .3mf save. Reload from disk fails with the same generic error:
```
Unable to reload:
<filename>
```

FrHePo, 2022-03-14:
Tried to 'reload from disk' (AMF-file) with the new version 2.4.1 (Windows 10, 64 bit). Result: still the same error as above!
I just don't get it. Can someone explain this to me? 
Even in the  case that metadata for the amf-(or other) file are missing or erroneous why not open a user dialogue to allow the user to enter the correct file?

FrHePo, 2022-03-14:
Another way to provoke this 'reload' error: 1. New project. 2. Load AMF-file. 3. Save project. 4. recall project. 5. Reload from disk --> above error.

FrHePo, 2022-03-16:
No comments? Nobody interested?

jmerc77, 2022-03-16:
one thing thats a bit odd is having replace with stl and reload from disk as separate functions internally. it would be more wise to just have 1 function for replace with file (any model not stl only) and for reload from disk simply call replace with file with the path saved in the project/project file. this way however, if one fails they both fail. but on the other hand, the failing is all in one place making it easy to fix.

FrHePo, 2022-03-16:
I agree. It seems to me that this important function just did not get the proper attention. I hope (and expect) that in the next beta version a good solution will be offered.

jmerc77, 2022-03-16:
do you mean 2.4.2? because 2.4.1 is now in release stage.

FrHePo, 2022-03-16:
Yes, 2.4.2 beta1 would be the first opportunity.

FrHePo, 2022-05-11:
2.4.2 Reload from disk bug again not addressed 🙈

FrHePo, 2022-08-12:
2.5.0-alpha3 nothing changed AFAIS 😭

jmerc77, 2022-08-13:
my finding at #7429 is still there pre 2.5.0. I must point out that its when I save project and reopen that it occurs. no issues before saving. haven't tested the findings in 2.5.0 but do have alpha3 installed. will update on this in #7429.

FrHePo, 2022-10-11:
Version 2.5.0 When you open a saved project and try to reload (an amf-file) from disk, you still get an "unable to reload".
I really wonder, why this problem is being ignored. It must be more than a year old.

bombela, 2022-10-11:
For me, it reloads fine when AMF/STL/STEP contain a single part on 2.5.0.

Also, the F5 shorcut for reloading only works in the 3D view, not the gcode view (there, it is just ignored). But you an still click through the menu.

FrHePo, 2022-10-11:
[image]
Right click -> reload from disk -> error

FrHePo, 2022-10-11:
Before the project is saved 'reload from disk' works fine but after saving, exiting and reloading the project 'reload from disk' produces the above error. This happens consistently on both my laptop and my workstation.

bombela, 2022-10-12:
I wonder if the style of path makes a difference. I notice you have a windows NT absolute path (`\\blabla`). What if you use a `C:\` style path?

jmerc77, 2022-10-12:
Does it exist? AFAIK that means that file went poof!(at some point)

jmerc77, 2022-10-12:
While I don't have that problem I see why.
BTW I also think that might be it. So ignore  my file gone like yesterday
explanation. BTW some may be noticing the puns. Just released from jail
over some stupid actions I did so I'm glad to be back at the
keyboard/cellphone. (And also is why the delay)

FrHePo, 2022-10-12:
> The style of path makes no difference. On my workstation the files lie on my local disk. The problem is the same.

FrHePo, 2022-12-11:
I removed all occurences of the Prusa slicer software. I really cleaned my laptop (Windows 10) of all Prusa software. Then I did a new install of Prusa slicer 2.5.0. Then I created a new project with an .amf-file. After saving the project, quitting, starting Prusaslicer again I recalled the project and tried to 'reload from disk'. As usual I got an error (see attached picture).
[image]
I tried both 'reload from disk'-export configuration options.
I really wonder why this bug is not addressed. It's easy to reproduce and is not dependent of any complex condition.

jmerc77, 2022-12-11:
Never got that error but had other issues. Maybe try on a fresh win10 install just in case your os is corrupt.

bombela, 2022-12-12:
That's not the OS. Or rather the chance of a bug in the operating system
impacting such a thing as re-opening a file is incredibly slim. It
technically possible. But it's like asking somebody to replace the whole
house because the refrigerator is not working. Sure it could be the
electrical that needs fixing. But it's very much more likely the
refrigerator itself.

jmerc77, 2022-12-12:
Come to think of it, the AMF reader could be the problem. Try converting to some other format and see if anything changes. Btw could you or someone translate the error to English please.

jmerc77, 2022-12-12:
the error seems to be shown in Plater.cpp-> `Plater::priv::reload_from_disk()` at the end of the function.
tracing back through the fail_list that triggers the error there are 4 if conditions all pushing to fail_list under the same condition of: `if(new_object_idx < 0 || int(new_model.objects.size()) <= new_object_idx)`...

given the nature of the condition, it seems that there needs to be no new volumes in the imported AMF file. not sure though. the question is what is the differences between the old AMF and the new AMF?

jmerc77, 2022-12-12:
adding the missing name did nothing. however, it does seem to be when opening from project file. which is similar to a problem i had with placement of stls in the slicer.

closed 2026-09-29 by prusaslicer-bot[bot]