prusa3d/PrusaSlicer #7590: Is there an eraser function for Paint-on supports?

laptop-laptop, 2021-12-24:
### Version
2.4.0

_Use `About->About PrusaSlicer` for release versions_
2.4.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3MK3S+

### Behavior
Sometimes I applied Paint-on support or Support Blockers on unintended area. Is there some kind of eraser function so that I can erase those blue or red marks?

How come sometimes when I moved the cursor on part of the model to pain it, the cursor did not have a circular shape for me to apply Paint-on Support or Support Blockers. Under what situations are we not able to use these functions?

_Is this a new feature request?_

Yes if it does not exist.

foreachthing, 2021-12-24:
It should be described in the tooltip. 
I _think_ you can erase while holding the shift
key. 

https://m.youtube.com/watch?v=RicHCOd7DqA

laptop-laptop, 2021-12-24:
Erasing is working. Thank you.

Anybody knows why sometimes the cursor does not change shape to allow Paint-on Support or Support Blockers?

n8bot, 2021-12-24:
Maybe because you have pressed the alt key, and the menus are highlighted waiting for input, and so the rest of the UI is locked? I notice in that case that the cursor will not re-size until I click anywhere in the 3d preview.

laptop-laptop, 2021-12-27:
I do not use the alt key. My impression seems to be that sometimes PrusaSlicer treats 3D object as a 2D plane especially when using the Paint-on-support, blocker support functions.

n8bot, 2021-12-27:
There is a circle and a sphere. The sphere is aware of 3D, and the circle just projects a circle from the direction the camera is looking at the object.

rtyr, 2022-01-03:
Answered. Closing.

closed 2022-01-03 by rtyr