prusa3d/PrusaSlicer #7592: "Send to printer" dialog UI change in 2.4 lacks checkbox/option that retains "Upload and print" by default

MikePfaff, 2021-12-24:
### Version
2.4.0+x64

### Operating system type + version
macOS 10.15.7 (19H1519)

### 3D printer brand / version + firmware version (if known)
Any printer

### Behavior
* _Steps to reproduce_
** Click "Send to printer" button
** Don't change anything in the dialog
** Hit "Enter" on the keyboard

* _Expected Results_ (= behavior before 2.4)
** Explanation about the dialog: After clicking the "Send to printer" button, the "Send to printer" dialog will open, with the checkbox for automatic start of printing already checked (because this checkbox retains its "checked" state from the previous print). Also the "Send to printer" (or was it labeled "OK"?) button is pre-selected and so can be easily triggered with the "Enter" key.
** File is sent to printer
** Print is automatically started 

* _Actual Results_ (= behavior in 2.4)
** Explanation about the dialog: After clicking the "Send to printer" button, the "Send to printer" dialog will open, but there is no checkbox for automatic start of printing anymore. Instead there are two buttons "Upload" and "Upload and print", but the "Upload" button is _ALWAYS_ pre-selected and therefore only the "Upload" function can be easily triggered with the "Enter" key. So to upload WITH automatic start of printing, now you need to use the mouse to actually click on the "Upload and print" button _on every print_. 
** File is sent to printer
** Print is NOT automatically started

_Is this a new feature request?_
No

dartrax, 2021-12-24:
Duplicate of #7564
Note that you can use Tab and Enter for Upload and Print. Still not as comfortable as before, I have to admit.

MikePfaff, 2021-12-24:
Hitting  "tab" to switch the focused button from "Upload" to "Upload and Print" in the dialog doesn't work at all on macOS. The only way to trigger "Upload and Print" on macOS is to click on the button with the mouse, which is rather tedious compared to "Just hit enter" (if you have in the past at some point checked the checkbox) in previous versions.

rtyr, 2022-01-03:
Closing as a duplicate.

closed 2022-01-03 by rtyr

MikePfaff, 2022-01-15:
@dartrax This is not a duplicate of #7564 as:

a) The issue here is about the loss of functionality due to the UI change in 2.4. The lost functionality in this case being "PrusaSlicer does not retain the preference of 'Upload and Print' for subsequent prints anymore".

b) Issue #7564 is about a problem when utilizing the UI with a keyboard. Even with #7564 fixed, you still need to choose between "Print" and "Upload and Print" every time. So this issue here will remain unsolved, as the preference of "Upload and Print" is still not retained for subsequent prints.

c) Even with #7564 fixed, the suggested "Just hit tab+enter every time" workaround for the "lost checkbox" will not work on MacOS unless you change your system-wide(!) preferences, which can have undesired effects on other applications on the system.

@rtyr I therefore kindly request that this issue is reopened.

ChrisHeerschap, 2022-02-01:
+1 on this request, I almost always started the print with the upload and can confirm @MikePfaff 's comments about tab not selecting the buttons on Mac OS. Have already sent two prints today only to wonder several minutes later why it wasn't printing yet. I can't think of any good reason to have taken away the ability to set a default to upload and print and echo the request to put it back.

bubnikv, 2022-02-01:
> b) Issue #7564 is about a problem when utilizing the UI with a keyboard. Even with #7564 fixed, you still need to choose between "Print" and "Upload and Print" every time. So this issue here will remain unsolved, as the preference of "Upload and Print" is still not retained for subsequent prints.

We received feedback that the old checkbox style dialog lead to unwanted starting of a print. Thus we removed the checkbox with its persistent behavior.

We never make everybody happy.

ChrisHeerschap, 2022-02-16:
So the people who can't be bothered to read win? Super.